arduino/ArduinoCore-renesas #3: Is license MIT?

NoMaY-jp, 2023-06-23:
Hello,

Yesterday, the LICENSE file was added. It is the MIT license. Is it right?

Best regards,
NoMaY

facchinm, 2023-06-23:
Exactly! :slightly_smiling_face:

NoMaY-jp, 2023-06-23:
Nice! 👍

I remember that FreeRTOS changed its license from \`Modified GPL license\` to the MIT license several years ago.

I think that this is the same impact.

NoMaY-jp, 2023-06-24:
Hello,

I've forked and cloned the repo and I'm looking files. I have a question. This repo has files which have the LGPL 2.1 license statements and non-corporate copyrights. For example:

https://github.com/arduino/ArduinoCore-renesas/blob/main/libraries/EEPROM/src/EEPROM.h
```
  Original Copyright (c) 2006 David A. Mellis.  All right reserved.
  New version by Christopher Andrews 2015.
  Original Copyright (c) 2006 David A. Mellis.  All right reserved.
  New version by Christopher Andrews 2015.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.
```

My understanding is that the files in the following folder are the MIT license. Is my understanding right?

https://github.com/arduino/ArduinoCore-renesas/tree/main/cores/arduino

Best regards,
NoMaY

NoMaY-jp, 2023-06-26:
Hello,

I'm looking files. The following folder contains files which have the LGPL 2.1 license statements and non-corporate copyright.

https://github.com/arduino/ArduinoCore-renesas/tree/main/cores/arduino/usb

For example:

https://github.com/arduino/ArduinoCore-renesas/blob/main/cores/arduino/usb/USB.cpp
```
    Copyright (c) 2021 Earle F. Philhower, III <earlephilhower@yahoo.com>
    This library is free software; you can redistribute it and/or
    modify it under the terms of the GNU Lesser General Public
    License as published by the Free Software Foundation; either
    version 2.1 of the License, or (at your option) any later version.
```
Best regards,
NoMaY

NoMaY-jp, 2023-06-27:
Hello,

I'm looking files in the cores/arduino folder. I have a question. Is there any reason that MIT License statements shouldn't be added to each file? 

I made a DOS batch file to add the statements to each file and push the updated files to my repo. For example:

Updated:

https://github.com/NoMaY-jp/ArduinoCore-renesas/blob/ArduinoCore-Renesas-explicit-MIT-License-Statements-in-files/cores/arduino/Arduino.h#L1-L25

Original:

https://github.com/arduino/ArduinoCore-renesas/blob/main/cores/arduino/Arduino.h

Best regards,
NoMaY

P.S.

The DOS batch file is the following:

```
set SRC=C:\Renesas\GitHubDesktop\workspaces\work_ArduinoCoreRenesas\branch_1_Add_MIT_licence_to_files_in_cores_arduino\1_org
set DST=C:\Renesas\GitHubDesktop\workspaces\work_ArduinoCoreRenesas\branch_1_Add_MIT_licence_to_files_in_cores_arduino\2_upd
set CAT=C:\Renesas2\MinGW\msys\1.0\bin\cat.exe

call :update analog.cpp
call :update analog.h
call :update api
call :update Arduino.h
call :update cortex_handlers.c
call :update dac.cpp
call :update dac.h
call :update digital.cpp
call :update FifoBuffer.h & rem TODO: Fix this file by hand. This file contains `Copyright (c) 2014 Arduino` and LGPL 2.1.
call :update FspLinkIrq.h
call :update FspTimer.cpp & rem TODO: Fix this file by hand. This file contains both CR and CRLF.
call :update FspTimer.h
call :update FspTransfer.h
call :update Interrupts.cpp & rem TODO: Fix this file by hand. This file contains `Copyright (c) 2014 Arduino` and LGPL 2.1.
call :update IRQManager.cpp
call :update IRQManager.h
call :update itoa.c & rem TODO: Fix this file by hand. This file contains `Copyright (c) 2014 Arduino` and LGPL 2.1.
call :update main.cpp & rem TODO: Fix this file by hand. This file contains both CR and CRLF.
call :update pinDefinitions.cpp
call :update pulse.cpp
call :update pwm.cpp
call :update pwm.h
call :update SafeRingBuffer.h & rem TODO: Fix this file by hand. This file contains `Copyright (c) 2014 Arduino` and LGPL 2.1.
call :update Serial.cpp & rem TODO: Fix this file by hand. This file contains `Copyright (c) 2018-2019 Arduino SA` and LGPL 2.1.
call :update Serial.h & rem TODO: ??? This file contains the following names and LGPL 2.1.
rem                               Copyright (c) 2006 Nicholas Zambetti.
rem                               Modified 28 September 2010 by Mark Sproul
rem                               Modified 14 August 2012 by Alarus
rem                               Modified 3 December 2013 by Matthijs Kooijman
call :update SerialObj1.cpp
call :update SerialObj2.cpp
call :update SerialObj3.cpp
call :update shift.cpp
call :update sync.h
call :update time.cpp
call :update Tone.cpp
call :update variant.h
call :update variant_helper.cpp
call :update vector_table.c
call :update WMath.cpp

goto :EOF

:update

"%CAT%" .\LICENSE.h %SRC%\%1 > %DST%\%1

goto :EOF
```

Best regards,
NoMaY

aentinger, 2023-08-01:
>I made a DOS batch file to add the statements to each file and push the updated files to my repo. For example:

Well, that's neat, but doesn't really help us. If there are missing license statements then please open a PR adding them.

I'll be closing this PR as the initial question has been answered.