arduino/ArduinoCore-renesas #10: Upload broken for the Uno R4 WiFi on Windows

SFE-Brudnerd, 2023-06-26:
Hello,

I'm having an issue with uploading to the Uno R4 WiFi in a Windows environment. 

This all uses ArduinoCore-renesas v1.0.1. When in the Arduino IDE, the device is recognized. It's only when uploading to it that it fails to find the device.

[image]

When looking at the device manager, I get the following values: USB VID 2341 PID 1002
[image]

Below is a table of the environments/configurations I tried and the various error codes I received.

Platform | IDE | Sketch | Success | Error Code
-- | -- | -- | -- | --
Mac | 2.1.0 | Blink | Y | N/A
Mac | 2.1.0 | GameOfLife | Y | N/A
Ubuntu | 2.1.0 | Blink | Y | N/A
Ubuntu | 2.1.0 | GameOfLife | Y | N/A
Windows 10 | 2.1.0 | Blink | N | Performing 1200-bps touch reset on serial port COM22<br>"C:\Users\\$USER\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino5/bossac" -d --port=COM22 -U -e -w "C:\Users\\$USER\AppData\Local\Temp\arduino\sketches\FE8CA22D18608F6CBD5B3FFBE2AB9B4C/Blink.ino.bin" -R <br> No device found on COM22<br> Set binary mode <br> Send auto-baud <br> Set binary mode<br> Failed uploading: uploading error: exit status 1
Windows 10 | 2.1.0 | GameOfLife | N | Performing 1200-bps touch reset on serial port COM22<br> "C:\Users\\$USER\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino5/bossac" -d --port=COM22 -U -e -w "C:\Users\\$USER\AppData\Local\Temp\arduino\sketches\63B0DA378C2FD7222A66F165BF61AF79/GameOfLife.ino.bin" -R<br> Set binary mode<br> Send auto-baud<br> Set binary mode<br> No device found on COM22<br> Failed uploading: uploading error: exit status 1
Windows 10 | Arduino-cli | Blink | N | Performing 1200-bps touch reset on serial port COM22<br> "C:\Users\\$USER\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino5/bossac" -d --port=COM22 -U -e -w "C:\msys64\tmp\arduino\sketches\FE8CA22D18608F6CBD5B3FFBE2AB9B4C/Blink.ino.bin" -R<br> Set binary mode<br> Send auto-baud<br> Set binary mode<br> No device found on COM22<br> Error during Upload: Failed uploading: uploading error: exit status 1
Windows 10 | VSCode Arduino | Blink | N | [Starting] Uploading sketch '..\..\..\..\Arduino\Blink\Blink.ino'<br> [Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.<br> Sketch uses 34828 bytes (13%) of program storage space. Maximum is 262144 bytes.<br> Global variables use 2460 bytes (7%) of dynamic memory, leaving 30308 bytes for local variables. Maximum is 32768 bytes.<br> No device found on COM22<br> Error during Upload: Failed uploading: uploading error: exit status 1<br> IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"<br> [Error] Uploading sketch '..\..\..\..\Arduino\Blink\Blink.ino': Exit with code=1

When I was reading through the [R4 Datasheet](https://docs.arduino.cc/static/28f8f808e387a0c85b122ad07d55bd60/ABX00087-datasheet.pdf) I see that the R4 is programmed via the ESP32-S3. There's a jumper I could solder to move the USB directly to the Renesas chip, but that seems like the wrong way to go about trying to program the device based on it's intended use.

I followed this [help article](https://support.arduino.cc/hc/en-us/articles/4407830972050-Find-and-stop-process-blocking-a-port) for the specific error, however the process explorer was unable to find a PID with the process in use.

I've been able to upload to the Minima using v1.0.1 in a Windows environment.

As a last ditch effort, I uninstalled Arduino, uninstalled the drivers, unplugged the device, rebooted the machine, reinstalled Arduino, installed the package, then plugged the device back in. This did not change anything.

Please let me know if there's something else I should be trying.

maxsrobotics, 2023-06-28:
I'm also having the exact same issue as described.

maxsrobotics, 2023-06-29:
Try Arduino 1.8. I was able to get it working there.

DanielTerletzkiy, 2023-06-29:
Same issue here.
Does anyone know when Platformio will add support for this board?

ubidefeo, 2023-06-30:
hey everyone
we just released a patch fix for Arduino IDE.
You can issue a "check for updates" command in the IDE and be notified to install 2.1.1

Sorry for the inconvenience, hope it'll all work for everyone now :)

qwikag, 2023-07-01:
> hey everyone we just released a patch fix for Arduino IDE. You can issue a "check for updates" command in the IDE and be notified to install 2.1.1 If you have issues with the internal updates just download the full build from our GitHub release section https://github.com/arduino/arduino-ide/releases/tag/2.1.1
> 
> Sorry for the inconvenience, hope it'll all work for everyone now :)

Why don't you advise what was wrong and how you fixed it?
was it really just a versioning typo

per1234, 2023-07-01:
Hi @qwikag. The problem was in the code in [**Arduino CLI**](https://arduino.github.io/arduino-cli/latest/) that signals the start of an upload process to the target board by momentarily opening a serial connection at 1200 bps (["1200 bps touch"](https://arduino.github.io/arduino-cli/dev/platform-specification/#1200-bps-bootloader-reset)).

You can see the change to the Arduino CLI codebase that was made to fix the problem here: https://github.com/arduino/arduino-cli/pull/2234

There is some further information about the change in that PR.

Once the problem was fixed in Arduino CLI, it was only necessary to update the version of Arduino CLI that is bundled with the Arduino IDE 2.x installation in order to also deliver the fix to Arduino IDE users: https://github.com/arduino/arduino-ide/pull/2121

> was it really just a versioning typo

No

qwikag, 2023-07-01:
> Hi @qwikag. The problem was in the code in [**Arduino CLI**](https://arduino.github.io/arduino-cli/latest/) that signals the start of an upload process to the target board by momentarily opening a serial connection at 1200 bps (["1200 bps touch"](https://arduino.github.io/arduino-cli/dev/platform-specification/#1200-bps-bootloader-reset)).
> 
> You can see the change to the Arduino CLI codebase that was made to fix the problem here: [arduino/arduino-cli#2234](https://github.com/arduino/arduino-cli/pull/2234)
> 
> There is some further information about the change in that PR.
> 

Thank you for the super fast response!
I will go look there, thank you.

balazscseh, 2023-07-01:
@per1234 The problem still persist in the VS Code Arduino extension, and i cannot see the 1200 bps touch update in the changelog.
Will this be fixed too? I don't see anyone mentioning this in the vsode-arduino repo, except @SFE-Brudnerd in his table.

per1234, 2023-07-01:
Hi @balazscseh. If you are referring to the ["**Visual Studio Code extension for Arduino**" extension](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino), it is already fixed. You only need to update [the installation of Arduino CLI that is used by the extension](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino#arduino-cli) on your computer to the latest version, which contains the fix I referenced above.

There is information about how to get the latest version of Arduino CLI here:

https://arduino.github.io/arduino-cli/latest/installation

balazscseh, 2023-07-01:
@per1234 
So i can somehow update the bundled Arduino-CLI? If yes, can you please explain me how to do it?

per1234, 2023-07-02:
@balazscseh I just gave ["Visual Studio Code extension for Arduino"](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) version 0.6.0 with the `arduino.path` setting pointing to my installation of Arduino CLI 0.33.1 and it works fine for me.

Please make a post with a detailed description of your problem on Arduino Forum. I'll be happy to investigate the problem further and try to help you over there. The forum a better place for that sort of user support level discussion:

https://forum.arduino.cc/

balazscseh, 2023-07-02:
I still get the same old error after pointing the path setting to the correct location.

My post link is: https://forum.arduino.cc/t/still-no-device-found-on-com-port-after-updating-cli-to-0-33-1/1143837

Thanks a lot :)

[image]

per1234, 2023-07-04:
This issue can be closed as resolved by:

- Arduino CLI: https://github.com/arduino/arduino-cli/pull/2234 (released in [0.32.3](https://github.com/arduino/arduino-cli/releases/tag/0.32.3) and [0.33.1](https://github.com/arduino/arduino-cli/releases/tag/0.33.1))
- Arduino IDE: https://github.com/arduino/arduino-ide/pull/2121 (released in [2.1.1](https://github.com/arduino/arduino-ide/releases/tag/2.1.1))
- Arduino Cloud: https://github.com/arduino/arduino-create-agent/pull/802 (released in [1.3.2](https://github.com/arduino/arduino-create-agent/releases/tag/1.3.2))

As for the report of the problem with the ["Visual Studio Code extension for Arduino" extension](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino), it is working fine for me. I will provide support on [the forum topic](https://forum.arduino.cc/t/still-no-device-found-on-com-port-after-updating-cli-to-0-33-1/1143837). Even if there is still some problem, I don't foresee that it will turn out to be in this repository's codebase (since all the work so far has been done in the Arduino CLI codebase) so I wouldn't consider that report a blocker for closing this issue.

CC: @facchinm @aentinger

aentinger, 2023-07-05:
Thank you @per1234 for leading the response on this as well as the neat summary. In light of it (your summary) I'll close this issue. I expect with all the fixes out fewer users should come looking for support and if - they can be simply redirected to the now existing answers.

closed 2023-07-05 by aentinger