arduino/ArduinoCore-renesas #12: SPI mode and bit order not set correctly

trevor-makes, 2023-06-28:
See [this line in SPI.cpp](https://github.com/arduino/ArduinoCore-renesas/blob/149f78b6490ccbafeb420f68919c381a5bdb6e21/libraries/SPI/SPI.cpp#L386C38-L386C38)

`spcmd0 &= !(((uint32_t)0xFF) << 2);` should be `spcmd0 &= ~(((uint32_t)0xFF) << 2);`

The logical not makes the mask 0 and wipes out the earlier settings. Should be a bitwise not to make the mask FFFFFC03.

trevor-makes, 2023-06-28:
Looking into this a bit further, I'm not sure why this would be the mask setting. BRDV is only two bits (b3, b2), so I think the mask should actually be `~(((uint32_t)0x03) << 2)`.

I'm experimenting with setting the registers directly, but haven't had any luck thus far.

trevor-makes, 2023-06-28:
Aha, got it working... I was using the SPI0 registers, but should have been using SPI1. Adding the following after calling `beginTransaction` set the expected LSB and MODE3 behavior, so fixing the mask as above should resolve the issue.

```
  R_SPI1->SPCMD_b[0].CPHA = 1;
  R_SPI1->SPCMD_b[0].CPOL = 1;
  R_SPI1->SPCMD_b[0].LSBF = 1;
```

aentinger, 2023-08-01:
Closing this issue since it is heavily outdated, see https://github.com/arduino/ArduinoCore-renesas/pull/45 .

closed 2023-08-01 by aentinger