arduino/ArduinoCore-renesas #15: Linux fails uploading with Failed uploading: uploading error: exit status 74 /

ccoenen, 2023-06-29:
I just received my Arduino UNO R4 Minima, and uploading to it fails. I am on Fedora Linux (38, updated) and have the Arduino IDE 2.1.0 installed from Flatpak.

I am trying to upload the standard blink example to it. It compiles fine, but uploading gives this result:

```txt
dfu-util: Cannot open DFU device 2341:0069 found on devnum 76 (LIBUSB_ERROR_ACCESS)
dfu-util: No DFU capable USB device available
Failed uploading: uploading error: exit status 74
```

My user usually has no problems uploading to any board or connecting via Serial. I am in the dialup group. I did _not_ set any particular udev-rules, though. If that's something that is expected of me, I would value a hint.

ccoenen, 2023-06-29:
I added these udev-rules, and I was able to upload **once** to my board:

```txt
# Arduino Uno R4 Programming needs these
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="2341", ATTR{idProduct}=="0069", TAG+="uaccess"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="2341", ATTR{idProduct}=="0369", TAG+="uaccess"
```

I am not particularly experienced in udev rules, so please don't blindly copy my non-solution. But if anyone has pointers for me what I should do differently, please let me know.

ccoenen, 2023-06-29:
I also added a mode to my udev rules. It now works **every other time**. In between it fails with:

```
dfu-util: Lost device after RESET?
Failed uploading: uploading error: exit status 74
```

Current udev-rules (use at your own peril!):
```
# Arduino Uno R4 Programming needs these
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="2341", ATTR{idProduct}=="0069", TAG+="uaccess", MODE="0666"
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="2341", ATTR{idProduct}=="0369", TAG+="uaccess", MODE="0666"
```

ccoenen, 2023-06-29:
(same behaviour in Arduino IDE 1.8.19, by the way)

ccoenen, 2023-06-30:
Interesting, that rule did not get created on my machine. Should this have been happening automatically (is this even possible in flatpak)?

ccoenen, 2023-06-30:
I can confirm that that udev rule generally works, but it also does the "fails every other time" problem i've had with mine ("lost device after reset"). If this is unexpected, I can turn this into its own issue, of course.

aentinger, 2023-08-01:
I'll be closing this due to inactivity, @ccoenen please reopen if this issue should persist on your end.

closed 2023-08-01 by aentinger