arduino/ArduinoCore-renesas #17: CMSIS-DSP

pschatzmann, 2023-06-30:
I was wondering if we can use functionality from CMSIS-DSP.
Any hints will be appreciated...

aentinger, 2023-07-03:
@karlsoderby this is probably rather something to be done in a tutorial I suppose?

@pschatzmann I leave this open for now, but I think this would have been a better fit for the forum.

pschatzmann, 2023-07-03:
So can you confirm that it is already included (eg. like in the rp2040 core) or do I need to install it as separate library like in the [Arduino Nano 33 BLE](https://store.arduino.cc/arduino-nano-33-ble) ? 
The [Arduino_CMSIS-DSP](https://github.com/arduino-libraries/Arduino_CMSIS-DSP/tree/master)  library is only working for the architectures=mbed,mbed_nano,mbed_portenta,mbed_rp2040,mbed_edge

aentinger, 2023-07-05:
I'm not sure the Arduino_CMSIS-DIP library you've referenced is doing what you think it should do. This brings me to the most important question: I believe I have not yet fully captured what you would like to do. Can you please rephrase what outcome you'd like to achieve?

pschatzmann, 2023-07-05:
I would like to use the FFT and the DSP Filtering functionality provided by the CMSIS DSP library.
Usually this library is made available to all ARM processors and provides some optimized DSP functionality.

aentinger, 2023-07-05:
As per my understanding I can confirm that CMSIS DSP library is not built into the static library `libfsp.a` that's part of ArduinoCore-renesas.

[image]

@facchinm what do you think of compiling CMSIS DSP into `libfsp.a`? [Arduino_CMSIS-DSP](https://github.com/arduino-libraries/Arduino_CMSIS-DSP/blob/master/create_lib) looks like a very custom solution made for mbed only.

facchinm, 2023-07-10:
Hi @pschatzmann ,
[Arduino_CMSIS-DSP](https://github.com/arduino-libraries/Arduino_CMSIS-DSP/tree/master) should work everywhere, the information in library.properties is outdated.
The library can also be downloaded from Library Manager and I just tested it to work out of the box :slightly_smiling_face:

closed 2023-07-10 by facchinm

pschatzmann, 2023-07-10:
Cool, thanks! 

I just had to adapt one include file in my library and the functionality was compiling.
Works like a charm: https://www.pschatzmann.ch/home/2023/07/10/arduino-uno-r4-fft-using-cmsis-dsp/