arduino/ArduinoCore-renesas #18: The strnlen() C Function Is Not Available

woolseyj, 2023-06-30:
It appears that several C functions, e.g. `strnlen()` and `strdup()`, defined in the AVR based cores are not available within the Renesas core.

Both of the following statements work on an Arduino Uno WiFi Rev2 board but the second statement does not work on the new Arduino Uno R4 WiFi board.

```
Serial.print("strlen = ");  Serial.println(strlen(string));
Serial.print("strnlen = ");  Serial.println(strnlen(string, 10));  // error: 'strnlen' was not declared in this scope
```

I came across this issue by trying to compile a sketch that uses the [PubSubClient](https://github.com/knolleary/pubsubclient) library.

Please see [The strnlen() C Function Is Not Available](https://forum.arduino.cc/t/the-strnlen-c-function-is-not-available/1142717) topic within the Arduino Forum for more information.

WestfW, 2023-08-03:
It looks like this is because the compile command line includes `-D_XOPEN_SOURCE`, which as I understand it causes quite an old set of API standards to be defined by the gcc include file structure...

There is a standard string.h with
size_t _EXFUN(strnlen,(const char *, size_t));

Hmm. It's inside conditionals:

#if __POSIX_VISIBLE >= 200809
size_t	 _EXFUN(strnlen,(const char *, size_t));
#endif
Alas, tracing why __POSIX_VISIBLE gets set to a particular value seems to be a non-trivial task (or at least requires tool knowledge that I don't have.)
SAMD compiles, which theoretically use the same toolchain, work fine.
(and indeed, R4 compiles have #define __POSIX_VISIBLE 199209, while SAMD has 200809)

Ahh. A bit of brute force says that __POSIX_VISIBLE gets set based on _POSIX_C_SOURCE, which defaults to 200809 unless _XOPEN_SOURCE is set less than 700. (in sys/features.h)

For some reason, the R4 build line includes a -D_XOPEN_SOURCE (SAMD doesn't define it at all.)

I don't know if that's intentional, or if someone thought that it means "Open Source" in the sense of OSSW. (I dont think that that's what it means!) (I can't find anything in the Arduino R4 core, the Arduino-api core, or the Renesas fsp code, that actually references _XOPEN_SOURCE

This does imply that the problem is in the #include files, and the function is actually present in the C library that is linked in. So a workaround is simply to provide an explicit prototype like:

extern "C" size_t _EXFUN(strnlen,(const char *, size_t));

facchinm, 2023-08-04:
@woolseyj @WestfW this should be tackled by #87

aentinger, 2023-08-08:
@woolseyj or @WestfW :coffee: :wave: 

Can you confirm this change works for you?

woolseyj, 2023-08-08:
@aentinger, I updated the platform.txt file with the -D_XOPEN_SOURCE=700 options and successfully tested my sketch using the PubSubClient library.

aentinger, 2023-08-09:
Hi @woolseyj ☕ 👋 thank you for taking the time to confirm this, I'll proceed with merging #87 .

closed 2023-08-09 by aentinger

woolseyj, 2023-09-08:
Confirmed to be working in production 1.0.4 tag.

jimemo, 2023-09-19:
> Confirmed to be working in production 1.0.4 tag.

@woolseyj thank you for tracking this down to resolution. I am still having the `'strnlen' was not declared in this scope; did you mean 'strlen'?` errors when using the [pubsubclient ](https://github.com/knolleary/pubsubclient) library. I'm wondering if I need to force a new version of renesas-ra ? I've created a new project (which I thought would install the latest renesas-ra). I also tried adding this to my platform.ini in platformio.
```
build_flags = 
    -D_XOPEN_SOURCE=700
```
Are there any additional steps that I need?

Fashion-Corp, 2023-09-19:
Dear James Emerson,

You can use this one here temporarily, if you do not necessarily NEED that
it must be included in the original library?

size_t strnlen(const char *s, size_t maxlen) {
    const char *end = memchr(s, '\0', maxlen);
    return end ? (size_t)(end - s) : maxlen;
}

This ought to work well. It has been tried other places before. Check it
out if it is the way you'd want it. You can create a #pragma message()
directive on it, to remind you that when the error has been corrected then
you'll comment it out.

Sincerely Yours,
David Svarrer

per1234, 2023-09-19:
Hi all. Discussion of fixing the problem in PlatformIO is off topic here. You are welcome to continue the discussion in an appropriate channel such as [Arduino Forum](https://forum.arduino.cc/) or the PlatformIO forum