arduino/ArduinoCore-renesas #22: UNO R4 Minima: CAN transmit/receive doesn't work

srolf, 2023-07-01:
I'm using a R4 Minima together with a NXP TJA1051 CAN high speed transceiver at 500 kBaud.

**It's not possible to receive or transmit any CAN frames.**

The TJA1051 I've cross-checked in my exiting CAN setup on an ESP32 and there it's properly working (there is also a Raspberry Pi2 with a CAN HAT plus a USB2CAN-FD connected to my PC on which SavvyCAN and a socketCAN application is running).

By using the Arduino example "CANWrite" incl. changed baud rate from 250 to 500 kBaud, the following error is shown in the terminal:

`CAN.write(...) failed with error code -60003`

srolf, 2023-07-02:
One addition which isn't directly changing the upper issue: If I check the HW used in the [CAN tutorial for the Minima](https://docs.arduino.cc/tutorials/uno-r4-minima/can), I'm starting to wonder about the used transceiver breakout board. At least the available variants of SN65HVD230 ICs on DigiKey are all only supporting 3.3V and none 5V as it would be required for the Minima. Looks little like the tutorial page is copied over from Portenta C33 which is 3.3V IO level and would work with the SN65HVD230.

aentinger, 2023-07-03:
Hi @srolf :coffee: :wave: 

I've been successfully using [this breakout board](https://www.skpang.co.uk/collections/can-bus-boards/products/can-bus-can-fd-breakout-board-5v-supply-and-5v-logic) with the Uno R4 Minima and WiFi. It's PHY driver is a [MCP2516](http://ww1.microchip.com/downloads/en/DeviceDoc/20005284A.pdf).

The error code you see (`FSP_ERR_CAN_TRANSMIT_NOT_READY` = `-60003`) indicates that the previous frame has not been sent. This can have a couple of reasons:

- No termination on CAN bus.
- Wrong pins connected to CAN_TX/RX
- Calling `write` too often, i.e. before the previous frame has been sent.

CAN-Pins on Minima:
```bash
#define PIN_CAN0_TX       (4)
#define PIN_CAN0_RX       (5)
```

aentinger, 2023-07-03:
I've also just re-checked it w/ my hardware, CAN on Minima works fine :+1: 

Another stumbling block could have been to change the bitrate to 500k at CANWrite, but not at CANRead.

srolf, 2023-07-03:
Thank you @aentinger for the confirmation, it was motivating me to continue the analysis.

There are now two observation:

The most silly one I did by the PIN assignment: Used 3,4 instead of 4,5... stupid me started counting not with "0"... even if I was looking at it 20 times and also tried to prevent to have interchanged TXD/RXD. Now I can send frames even at 500 kBaud.

But there is a second one: Is the default frame format "extended" instead of "standard" frame format (29 vs 11 bit identifier)? If I use 0x123 as ID in the code, SavvyCAN is showing it as 0x00000123 and marks it as "Extended".

I'm not sure, but by ~/.arduino15/packages/arduino/hardware/renesas_uno/1.0.1/cores/arduino/api/CanMsg.h and

`CanMsg(uint32_t const can_id, uint8_t const can_data_len, uint8_t const * can_data_ptr)`

it's not looking like to be possible to and a parameter for extended or standard frame format.

aentinger, 2023-07-05:
>The most silly one I did by the PIN assignment: Used 3,4 instead of 4,5... stupid me started counting not with "0"... even if I was looking at it 20 times and also tried to prevent to have interchanged TXD/RXD. Now I can send frames even at 500 kBaud.

Well I'm glad it was nothing more serious :wink: .

Re standard/extended: you're right. Currently only extended IDs are supported. That being said the 32-bit ID variable has room for three additional bits which I would like to model very much like [linux/can.h](https://github.com/rhyttr/SocketCAN/blob/master/kernel/2.4/include/linux/can.h#L27-L29). So if the MSBit is set to 1, you've received/are sending a frame with extended ID (which is currently ignored), if its 0 you've received/are sending a standard ID.

Personally I'd like to add support for standard IDs and you are not the first one to ask. Alas I need internal support in order to start development or an external user provided PR that I can help getting merged. So if you want this feature at the very least create an issue with a feature request or - if you're feeling up to it - get started on a PR that I can help you along :wink: .

srolf, 2023-07-05:
> That being said the 32-bit ID variable has room for three additional bits which I would like to model very much like [linux/can.h](https://github.com/rhyttr/SocketCAN/blob/master/kernel/2.4/include/linux/can.h#L27-L29).

I see, thank you.
Now I also understood why only EFF and no SFF frames were received: The current driver applies 0x1FFFFFFFU as filter.

aentinger, 2023-07-05:
Good, I'll be closing this issue then. 

Please consider adding a feature request via another issue or start a PR on supporting standard CAN IDs :wink:

closed 2023-07-05 by aentinger