arduino/ArduinoCore-renesas #23: OPAMP example?

metanav, 2023-07-02:
How can we configure and use OPAMP? I could not find any code related to the OPAMP peripheral in the core. Only there are a few defines in the variants directory.

maxgerhardt, 2023-08-06:
This Arduino core was compiled without the "OPAMP" component of the underlying Renesas FSP framework. However, you 
1. Can still add those missing files (https://github.com/renesas/fsp/tree/v4.0.0)
2. Use direct writes to the OPAMP peripheral registers

I just copied the missing files next to the sketch and created a working OPAMP example for the Uno R4 WiFi, whose OPAMP pins are exposed on the analog pins A1 (Plus), A2 (Minus) and A3 (Output). I successfully tested a simple voltage follower example. 

See https://github.com/maxgerhardt/uno-r4-opamp

closed 2023-09-15 by aentinger

Fashion-Corp, 2023-09-15:
> completed via #97 <https://github.com/arduino/ArduinoCore-renesas/pull/97>

Dear Alexander Entinger et al,

I am a software development professional with 41 calendar-years of
experience, hereof somem 15 years in embedded and microprocessors, ranging
since the Z80, and up till todays date.

I am CTO in SkyFi Energy Aps of Denmark and CEO of Rational Intuitive Ltd.
of Kenya. My question is:

I am heading a team of quite young developers with maybe 2 to 5 years of
experience, mainly on the Web/Fullstack side of it.

We have decided on developing a product which will be using the Arduino UNO
R4, due to its versatility, Bluetooth, WiFi, CAN-bus, RTC etc., and I would
therefore like to understand how I can contribute from time to time on the
development of the software residing on GitHub.

Sincerely Yours,
David Svarrer

PR #97 (2023-09-15): Add OPAMP library for Uno R4 (WiFi, Minima)
Resolves #23 

Adds an OPAMP library for the Arduino Uno R4 WiFi and Minima boards. Portenta C33's chip does not have an OPAMP.

Does not rely on Renesas FSP code, writes to OPAMP registers directly, to save codesize.

Only supports starting the OPAMP, none of the fancier modes for AGC links / trigger inputs / start ADCs. Very simple.

Verified to be working on hardware using a MAX038-based function generator, Hantek 6022BL 2-channel oscilloscope and the basic "voltage follower" and "non-inverting amplifier" with gain 2 circuits. Output is channel 1  [... cut]