arduino/ArduinoCore-renesas #25: Insufficient initialization of SCI

mukyokyo, 2023-07-03:
Since you are using fsp, I would recommend using R_SCI_UART_BaudCalculate to minimize the baudrate error.
Currently, the baudrate is initialized with a large error rate, and the R4 is not able to handle the baudrate that the AVR was able to handle.

mukyokyo, 2023-07-03:
I see you are using R_SCI_UART_BaudCalculate.
I was premature. I'm sorry.

I could not communicate with the device communicating at 1Mbps, so I tried to capture the transmitted data with another serial monitor separately and got bit corrupted data.
I will try to measure the error rate in more detail.

I just noticed that err_rate is supposed to handle 1000 times the value, but I thought that if err_rate=5, it would be considered 0.005%.

aentinger, 2023-07-03:
Please let us know what you find, meanwhile I'll close this issue as it's root cause was a misunderstanding :wink:

facchinm, 2023-07-03:
Hi @mukyokyo ,
if you change line https://github.com/arduino/ArduinoCore-renesas/blob/main/cores/arduino/Serial.cpp#L275 to 5000 is the situation getting better?

mukyokyo, 2023-07-03:
> Hi @mukyokyo , if you change line https://github.com/arduino/ArduinoCore-renesas/blob/main/cores/arduino/Serial.cpp#L275 to 5000 is the situation getting better?

It did not change when we actually set 3000 assuming 3%.
From the results, the actual baud rate will not match the actual baud rate in the 250k..1.5Mbps range.
Since R_SCI_UART_BaudSet is being performed without looking at the result of R_SCI_UART_BaudCalculate, I suspect that an inconvenient value may have been adopted.
I will put in the debug code and see what values are actually set in the registers.
Worst case scenario, I will also try to get a follow-up from someone in Renesas.

mukyokyo, 2023-07-03:
I will just note what I have learned somewhat.

It is clear from the return value of R_SCI_UART_BaudCalculate that the erroneous handling of the digits in err_rate.
It must be specified in permil.

Second, even if R_SCI_UART_BaudCalculate returns FSP_SUCCESS, it does not appear that R_SCI_UART_BaudSet sets the baud rate to the expected baud rate.

Finally, the previous list of baud rates only describes the behavior at those values.
In fact, it was found experimentally that setting the baud rate to a certain range does not result in an unexpected baud rate, but rather an incorrect baud rate when a specific baud rate is specified.

The results are somewhat troublesome so far.

mukyokyo, 2023-07-03:
Since the only way to determine whether the actual baud rate is correct or not is to suspect the value held in the variable.
I dumped the value calculated by R_SCI_UART_BaudCalculate as follows, and it seems that the actual baud rate is halved when mddr=0.

```js
    static int ind = 0;
    const uint32_t baud_list[]={9600,57600,115200,230400,250000,460800,500000,600000,700000,800000,900000,1000000,1500000,2000000,3000000,4000000,6000000};
    fsp_err_t err;
    const bool bit_mod = true;
    const uint32_t err_rate = 3000;  // Meaning 3%
    baud_setting_t uart_baud;

    if (++ind > (sizeof(baudlist) / sizeof(uint32_t) - 1)) ind = 0;
    err = R_SCI_UART_BaudCalculate(baud_list[ind], bit_mod, err_rate, &uart_baud);
    if(err != FSP_SUCCESS) {
      Serial.println("BaudCalculate error");
    } else {
//      err = R_SCI_UART_BaudSet(&uart_ctrl, (void *) &uart_baud);
      Serial.print(" bgdm=");
      Serial.println(uart_baud.semr_baudrate_bits_b.bgdm);
      Serial.print(" abcs=");
      Serial.println(uart_baud.semr_baudrate_bits_b.abcs);
      Serial.print(" abcse=");
      Serial.println(uart_baud.semr_baudrate_bits_b.abcse);
      Serial.print(" cks=");
      Serial.println(uart_baud.cks);
      Serial.print(" brr=");
      Serial.println(uart_baud.brr);
      Serial.print(" mddr=");
      Serial.println(uart_baud.mddr);
      Serial.print(" semr_baudrate_bits=");
      Serial.println(uart_baud.semr_baudrate_bits);
    }
```

I tentatively tried "const bool bit_mod = false" and now I can generally communicate with the list of baud rates in the previous source (I haven't looked at the actual error rate).

mukyokyo, 2023-07-04:
The most likely possibility is a bug in R_SCI_UART_BaudCalculate.
I tried a simple program in e2studio on a bench with a different MPU, and the actual baudrate is halved by setting a specific baudrate.

Since the condition has been identified, the fix is a simple matter. However, since we will not be taking care of fsp here, we will pass it to the fsp side.
I raised the issue with my contact at renesas.

Finally, I leave it here to note that the baudrate error rate must be specified in multiples of 1000.

thanks

mukyokyo, 2023-07-04:
I'll offer a cute solution that won't be affected by the fsp update. I have actually confirmed that the problem goes away.
```js
  err = R_SCI_UART_BaudCalculate(baudrate, bit_mod, err_rate, &uart_baud);
  if (uart_baud.mddr==0) err = R_SCI_UART_BaudCalculate(baudrate, false, err_rate, &uart_baud);
  err = R_SCI_UART_Open (&uart_ctrl, &uart_cfg);
```

Basically I expect bitrate_modulation to be true, so please don't think that setting it to false all the time will solve the problem.

aentinger, 2023-07-05:
Hi @mukyokyo :coffee: :wave: 

Thank you for your investigation as well as sharing the results. Would you mind to prepare a PR for it?

mukyokyo, 2023-07-05:
The real purpose is different.
In the course of investigating the high number of dropouts of received data at high baudrate (over 200 kbps), I incidentally discovered that a baud rate mismatch was occurring.
When I am using my own send/receive routines with e2studio projects, I do not experience any overflow of received data even at several Mbps.
I felt that this fsp-dependent send/receive routine in the arduino library needs to be seriously modified.

In the meantime, I will leave the PR of this baudrate flaw to you.
I think the fsp team has been informed about the R_SCI_UART_BaudCalculate issue.

thanks

facchinm, 2023-07-05:
Hi @mukyokyo ,
I'd apply your patch anyway giving the proper attribution (so it would be great if you can prepare a PR), than we can tackle the overflow issues.
Please keep in mind that arduino APIs are supposed to be called by anyone in any context, so advanced but dangerous functionalities like DMA can't be used directly.

mukyokyo, 2023-07-05:
> Please keep in mind that arduino APIs are supposed to be called by anyone in any context, so advanced but dangerous functionalities like DMA can't be used directly.

The uart HAL/Common Stacks was set in this library, but it assumes that DTC will be used (which doesn't seem to work in practice).
If you don't intend to use DMA, having this set up conflicts with your requirements.

The commonly used baudrate of 115.2 kbps or lower is still reasonably functional in its current state. Am I the only one who is not satisfied with this level of baudrate?

mukyokyo, 2023-07-08:
As for the missing received data, it is due to the interrupt priority, as I suspected.
I am not sure how IRQManager.h should be handled in Arduino, but changing UART_SCI_PRIORITY from 12 to 11 will reduce the overflow even at high baudrate.

SCI generates an interrupt for every byte, so at high baud rates the load is very high.
If SCI-dependent applications require more performance in the future, the load will become higher and higher, and it may be difficult to cooperate with other applications.

I am not involved in the design of this library, but it is hard to come up with a good idea if the use of DMA is restricted.

aentinger, 2023-08-01:
Suggestions by @mukyokyo worked into https://github.com/arduino/ArduinoCore-renesas/commit/7b25c02a721b6120fb60535a53664bc5a0b154b5 by @facchinm .

closed 2023-08-01 by aentinger