arduino/ArduinoCore-renesas #29: Sources for Uno R4 Minima/WiFi bootloaders (SANTIAGO)

rexut, 2023-07-04:
Sadly the referred Git repository at

1. https://github.com/arduino/ArduinoCore-renesas/blob/fbb44b6a9a6e0e5fbbc3480a8f3f6a5237de3c67/bootloaders/SANTIAGO/README.md?plain=1#L3
2. https://github.com/arduino/ArduinoCore-renesas/blob/fbb44b6a9a6e0e5fbbc3480a8f3f6a5237de3c67/bootloaders/SANTIAGO/README.md?plain=1#L10

is not accessible.

**Where (or how) can I clone the source code?**

facchinm, 2023-07-04:
Hi @rexut ,
we are in the process of upstreaming our patches to tinyusb https://github.com/hathach/tinyusb/pull/2052 , as soon as that PR is merged we'll provide the board support and the source code for the bootloaders :slightly_smiling_face:

rexut, 2023-07-04:
Thanks @facchinm for the quick reply, this is good news and solves my wondering.

closed 2023-07-04 by rexut