arduino/ArduinoCore-renesas #36: [Feature Request] UNO R4: Support for CAN Standard Frame Format (in addtion to EFF)

srolf, 2023-07-05:
Please add in the UNO R4 CAN library the support for transmit/receive of CAN frames also in SFF.

Preferred would be an own parameter in the message creation which allows to define either SFF (maybe the default) or EFF.

---
Adding a hint towards the current CAN tutorials would be helpful, because it's not directly obvious to be forced/limited to EFF.
---
CAN 2.0A defines SFF.
CAN 2.0B defines EFF.
Most of the available MCP2515 libs are supporting SFF and EFF.
CAN on Arduino Due by "due_can" lib supports SFF and EFF.
In the automotive area SFF for communication is very common.

Lesani, 2023-07-21:
@aentinger please please please

designer2k2, 2023-07-21:
+1 also from me for this!

The R4 is sitting now idle due to this, for me in automotive things its useless with only extended frame support.

Is anybody working on this? Or is there information around on how to add it?

aentinger, 2023-07-24:
Your feature request(s) has been noted. I'll try to get some time scheduled for extending software-support for 11-Bit IDs.

srolf, 2023-07-24:
> Your feature request(s) has been noted. I'll try to get some time scheduled for extending software-support for 11-Bit IDs.

Thank you very much!

Out of curiosity: Maybe not possible, but I'm wondering if the CAN TXD/RXD pins can be remapped or are they "hard-wired"? Background: I was thinking about using a Minima and the [Ethernet Shield Rev2](https://store.arduino.cc/products/arduino-ethernet-shield-2) to create a gateway CAN<->ETH, but without remapping, but I would expect issues with the SD-card reader on that shield.

aentinger, 2023-07-24:
> Maybe not possible, but I'm wondering if the CAN TXD/RXD pins can be remapped or are they "hard-wired"?

They can also be remapped to (some) other pins, the key is to search [pinmux.inc](https://github.com/arduino/ArduinoCore-renesas/blob/main/variants/MINIMA/pinmux.inc) for `PIN_CAN_TX `and `PIN_CAN_RX`. However, as the CAN objects are instantiated in the library (i.e. [here](https://github.com/arduino/ArduinoCore-renesas/blob/main/libraries/Arduino_CAN/src/R7FA4M1_CAN.cpp#L328)) you'd need to change the library to change the pin-mapping.

srolf, 2023-07-24:
> you'd need to change the library to change the pin-mapping

Got it, thanks. I should work on my C++ skills... an alternative begin() would be helpful.

Lesani, 2023-07-24:
> > you'd need to change the library to change the pin-mapping
> 
> Got it, thanks. I should work on my C++ skills... an alternative begin() would be helpful.

You should make a separate Feature Request to keep the two topics clean. In my use case I don't need different pins, but I agree, if the hardware can do it, it should be configurable somehow.

Lesani, 2023-07-24:
> Your feature request(s) has been noted. I'll try to get some time scheduled for extending software-support for 11-Bit IDs.

Thank you!

aentinger, 2023-07-24:
> Got it, thanks. I should work on my C++ skills... an alternative begin() would be helpful.

@srolf please create a separate feature request.

aentinger, 2023-08-02:
Heads up everyone, mixed mode (supporting both 11-Bit standard CAN IDs and 29-Bit extended CAN IDs) is now being supported for Arduino Uno R4 `WiFi` and `Minima`. If you want to take a sneak preview take a look at https://github.com/arduino/ArduinoCore-renesas/pull/79 .

srolf, 2023-08-02:
Thank you, a Minima with TJA1051 and tiny OLED is still standing here :-)

PR #79 (2023-09-04): [Arduino_CAN] support both 11-Bit (standard) and 29-Bit (extended) IDs.
This fixes #36.

Requires https://github.com/arduino/ArduinoCore-API/pull/194 .
closed 2023-09-04 by aentinger