arduino/ArduinoCore-renesas #41: Hardware-Accelerator Crypto

maxgerhardt, 2023-07-08:
On the Uno R4 WiFi, I'd like to make use of the Renesas' chips SCE (Secure Cryptography Engine). According to the .h file, the chip does that peripheral.

https://github.com/arduino/ArduinoCore-renesas/blob/80faaf868603ff96ac3ab8c354331af965496718/variants/UNOWIFIR4/includes/ra/fsp/src/bsp/mcu/ra4m1/bsp_feature.h#L97C1-L99

Creating a new `.c` file and writing a small test program

```
#include <r_sce_if.h>

void test() {
    const char* msg = "Hello";
    sce_sha_md5_handle_t ctx;
    uint8_t sha256_out[256 / 8];
    uint32_t digestLen = 256/8;
    HW_SCE_McuSpecificInit();
    HW_SCE_Sha256Init(&ctx);
    HW_SCE_Sha256Update(&ctx, (uint8_t*)msg, strlen(msg));
    HW_SCE_Sha256Final(&ctx, sha256_out, &digestLen);

    printf("Hash: ");
    for(int i=0; i < 256/8; i++) printf("%02x", sha256_out[i]);
    printf("\n");
}
```

fails at the linking stage with
```
main.c:(.text.test+0xe): undefined reference to `HW_SCE_Sha256Init'
main.c:(.text.test+0x20): undefined reference to `HW_SCE_Sha256Update'
main.c:(.text.test+0x2a): undefined reference to `HW_SCE_Sha256Final'
```

The datasheets very unhelpfully say

[image]

And the implementation of `HW_SCE_Sha256Init()` and related is nowhere to be found in the FSP (neither v4.0 that you used for this core nor the current version).

Am I blind or do I have to sign a freaking NDA with Renesas to use the crypto accelerator hardware in a chip I already paid money for?

facchinm, 2023-07-10:
Hi @maxgerhardt ,
the R4 only supports AES but for some reason the symbols are declared with a `Sub` suffix :detective: 
```

maxgerhardt, 2023-07-10:
You're right, I misread "GHASH" as general hash, but it's 'just' the hash function used in the AES-GCM mode. Also.. it only seems to have the GCM mode, not an other mode (ECB, CBC, ...). That's really weird.

In any case, I think I should file this in the FSP repo first, then we can maybe update here. It'd be nice to take full advantage of the crypto acceleration this chip **does** have. Will reopen as needed.