arduino/ArduinoCore-renesas #50: Pin Definitions Conflict with Sketch/Library Variables

mjs513, 2023-07-14:
I posted this as well on the forum but may be better suited here: https://forum.arduino.cc/t/pin-definitions-conflict-with-sketch-library-variables/1147341

Many libraries (such as Eigen) and sketches may use variables such as A0-A5 and D0-D15 as variables versus pins definitions in pins_arduino.h:
```

#undef A0
#undef A1
#undef A2
#undef A3
#undef A4
#undef A5
static const uint8_t A0 = PIN_A0;
static const uint8_t A1 = PIN_A1;
static const uint8_t A2 = PIN_A2;
static const uint8_t A3 = PIN_A3;
static const uint8_t A4 = PIN_A4;
static const uint8_t A5 = PIN_A5;

// Digital pins
// -----------
#undef D0
#undef D1
#undef D2
#undef D3
#undef D4
#undef D5
#undef D6
#undef D7
#undef D8
#undef D9
#undef D10
#undef D11
#undef D12
#undef D13
#undef D14
#undef D15
#define D0  (0u)
#define D1  (1u)
#define D2  (2u)
#define D3  (3u)
#define D4  (4u)
#define D5  (5u)
#define D6  (6u)
#define D7  (7u)
#define D8  (8u)
#define D9  (9u)
#define D10 (10u)
#define D11 (11u)
#define D12 (12u)
#define D13 (13u)
#define D14 (14u)
#define D15 (15u)
```
So if you run a simple sketch:

```
void setup() {
  Serial.begin(115200);
  while(!Serial && millis() < 5000) {}
  int D0 = 1;
  Serial.println(D0);
  //digitalWrite(D0, HIGH);
}

void loop() {}
```

it will throw an error:
```
C:\Users\Merli\AppData\Local\Temp\.arduinoIDE-unsaved2023612-22120-liswj5.0ensn\sketch_jul12a\sketch_jul12a.ino: In function 'void setup()':
C:\Users\Merli\AppData\Local\Temp\.arduinoIDE-unsaved2023612-22120-liswj5.0ensn\sketch_jul12a\sketch_jul12a.ino:4:12: error: lvalue required as left operand of assignment
   int D0 = 1;
            ^
exit status 1
Compilation error: lvalue required as left operand of assignment
```

A more complex example would be using Eigen:
In file included from

C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Arduino.h:125:0,
                 from C:\Users\Merli\AppData\Local\Temp\arduino\sketches\019E3DD99F1ECF6C34FCED459E6CC679\sketch\eigen_example.ino.cpp:1:
d:\Users\Merli\Documents\Arduino\libraries\eigen-main\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h: In member function 'void Eigen::internal::lhs_process_one_packet<nr, LhsProgress, RhsProgress, LhsScalar, RhsScalar, ResScalar, AccPacket, LhsPacket, RhsPacket, ResPacket, GEBPTraits, LinearMapper, DataMapper>::operator()(const DataMapper&, const LhsScalar*, const RhsScalar*, ResScalar, Eigen::Index, Eigen::Index, Eigen::Index, Eigen::Index, Eigen::Index, Eigen::Index, int, Eigen::Index, Eigen::Index, Eigen::Index, Eigen::Index, Eigen::Index)':
C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\variants\MINIMA/pins_arduino.h:56:16: error: lvalue required as unary '&' operand
 #define D0  (0u)
                ^
d:\Users\Merli\Documents\Arduino\libraries\eigen-main\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1270:73: note: in expansion of macro 'D0'
           peeled_kc_onestep(1, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
                                                                         ^~
........
only one many errors. To correct this pins_arduino.h can be changed to (same way you did it for the analog pins):
```

// Digital pins
// -----------
#define PIN_D0   (0u)
#define PIN_D1   (1u)
#define PIN_D2   (2u)
#define PIN_D3   (3u)
#define PIN_D4   (4u)
#define PIN_D5   (5u)
#define PIN_D6   (6u)
#define PIN_D7   (7u)
#define PIN_D8   (8u)
#define PIN_D9   (9u)
#define PIN_D10   (10u)
#define PIN_D11   (11u)
#define PIN_D12   (12u)
#define PIN_D13   (13u)
#define PIN_D14   (14u)
#define PIN_D15   (15u)

#undef D0
#undef D1
#undef D2
#undef D3
#undef D4
#undef D5
#undef D6
#undef D7
#undef D8
#undef D9
#undef D10
#undef D11
#undef D12
#undef D13
#undef D14
#undef D15
static const uint8_t D0 = PIN_D0;
static const uint8_t D1 = PIN_D1;
static const uint8_t D2 = PIN_D2;
static const uint8_t D3 = PIN_D3;
static const uint8_t D4 = PIN_D4;
static const uint8_t D5 = PIN_D5;
static const uint8_t D6 = PIN_D6;
static const uint8_t D7 = PIN_D7;
static const uint8_t D8 = PIN_D8;
static const uint8_t D9 = PIN_D9;
static const uint8_t D10 = PIN_D10;
static const uint8_t D11 = PIN_D11;
static const uint8_t D12 = PIN_D12;
static const uint8_t D13 = PIN_D13;
static const uint8_t D14 = PIN_D14;
static const uint8_t D15 = PIN_D15;
```

this gets rid of the error messages in the same sketch and when using the Eigen Library for instance.  I did test eigen with the recommended changes and it works.

aentinger, 2023-07-17:
>Many libraries (such as Eigen) and sketches may use variables such as A0-A5 and D0-D15 as variables versus pins definitions in pins_arduino.h:

Hi @mjs513 :coffee: :wave: 

How many libraries does that actually affect? Do you know of other libraries? I suggest to keep things in perspective :wink: .

That being said your desired change seems reasonable to me (I'd replace `const` with `constexpr` though). Can you please prepare a PR so that we may run it against CI as well as so that other members of @arduino/arduinocore-developers can chime in?

mjs513, 2023-07-17:
@aentinger 
"constexpr" does not work with the R4.  Getting this error when I try using it.  According to what I see its available from c++11 only

`pins_arduino.h:75:8: error: unknown type name 'constexpr'
`

aentinger, 2023-07-18:
How about you prepare a PR so that we can jointly take a look? :pray: :wink: 

For info how to do so, see [here](https://github.com/arduino/ArduinoCore-renesas/issues/49#issuecomment-1636900108).

mjs513, 2023-07-18:
@aentinger  -  PR Submitted.

mjs513, 2023-07-21:
Going to close this issue since its taken care of with the PR.

closed 2023-07-21 by mjs513