arduino/ArduinoCore-renesas #61: Stanrdard CAN IDs not supported

Lesani, 2023-07-20:
I use Arduinos in an industrial context to communicate with and get info from our large scale machinery.

Up until now I used the SEED Studio Can shield, but have an issue with the size. I was quite irritated when I tried the Minima and Wifi boards, and only extended IDs are supported. Our CAN IDs are "standard" 11 Bit. My programming doesn't go as far as implementing that with a PR, please please (urgently) amend your library with support for standard IDs. I wasn't expecing such a limitation and already designed all my housings around a single uno sized board without a shield, and am now more or less standing in a "rue de la gack" with this situation.

srolf, 2023-07-24:
I'm sorry to see you struggling with the same limitation I did. If you are in an urgent need to proceed and require a small Arduino-like footprint plus CAN plus WiFi, maybe an ESP32 is an alternative in the form of a Wemos R1 D32 or Keystudio ESP32 Plus (there are cheaper ones around, but with some from AZ Delivery I observed issues with CAN in the past).

aentinger, 2023-07-24:
@Lesani your feature request has been noted, I'll be closing this issue due to being a duplicate of #36.

closed 2023-07-24 by aentinger

Lesani, 2023-07-24:
> I'm sorry to see you struggling with the same limitation like I did. If you are in an urgent need to proceed and require a small Arduino-like footprint plus CAN plus WiFi, maybe an ESP32 is an alternative in the form of a "Wemos D1 R32 (original, no clone)" or "Keyestudio ESP32 Plus" (there are cheaper ones around, but with some from AZ Delivery I observed issues with CAN in the past). But those are using 3.3V signal level and not 5V.

Thanks! If Arduino's stance would be "will not implement" then I'd consider it, but since @aentinger is working on it, I'll wait as to not have another uC type in the fleet. The only time I'd consider another change would be if one had the transceiver onboard, too bad neither Minima nor Wifi have one. Maybe some 3rd party ones will come to the market that include a transceiver