arduino/ArduinoCore-renesas #63: Changing Compiler Optimization Seems to have no affect.

mjs513, 2023-07-21:
In platforms.txt I changed the optimization level from -Os to -O2 and appears to have no affect performance.  Using a coremark sketch that I used before the issue becomes apparent.   See this post: https://forum.arduino.cc/t/changing-compiler-optimizations-had-no-affect/1150181/4.  In summary:

```
Uno R4 WiFi (48Mhz, -0s) -  81.81
Uno R4 WiFi (48Mhz, -02) -  81.81
UnoR4 Minima (-02) - 106.75
UnoR4 Minima (-0s) - 106.75
```
but when I change it for other boards:
```
Teensy 3.2 (48Mhz -0s)  -  86.23
Teensy 3.2 (48Mhz -02)  - 116.07
Feather M0 (48Mhz -0s)  -   48.61
Feather M0 (48Mhz -02)  -   83.85
CH32V307 Eval(48Mhz -0s) - 91.80
CH32V307 Eval(48Mhz -02) - 116.52
```
it appears to make a difference.  Is there something I am missing or can this be related to the micros issue.  From what I saw the time is calculated from using millis()

REF: https://github.com/PaulStoffregen/CoreMark

RudolphRiedel, 2023-07-22:
No idea about the coremark, but I just changed the three -Os it in AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.1\platform.txt to -O2 and -O0 and checked the timing with a logic analyzer.

Time between two SPI.transfer(data):
-Os: 11.064µs
-O0: 13.016µs
-O2: 10.828µs

This was in Atduino IDE 1.8.19.
So the setting does make a measureable difference.
You did not by any chance use "-0s" and "-02" as you posted instead of "-O2"?

I would argue though that -O2 should be the default setting with anything ARM, the size of the binary is generally not an issue with these.

mjs513, 2023-07-22:
@RudolphRiedel 
Pretty sure I used -O2 and not 02.  But redid it this morning making sure I closed the IDE first so it would reload the platform.txt changes.  This time it worked:
```
MINIMA
CoreMark 1.0 : 81.80 / GCC7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204] (flags unknown) / STACK
CoreMark 1.0 : 106.73 / GCC7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204] (flags unknown) / STACK

WIFI
CoreMark 1.0 : 81.81 / GCC7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204] (flags unknown) / STACK
CoreMark 1.0 : 106.74 / GCC7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204] (flags unknown) / STACK
```

Going to close this out as a none issue.

PS.  I do agree with you that the default optimization should be -O2.  They probably should have made it selectable but that is another story.

Thanks for the verification

closed 2023-07-22 by mjs513