arduino/ArduinoCore-renesas #64: Robotis Libraries DynamixelShield and Dynamixel2Arduino examples do not build (or work)

KurtE, 2023-07-24:
As I mentioned on the forum (https://forum.arduino.cc/t/robotis-dynamixel-shield-code-does-not-compile/1150975)

The Robotis libraries which are used with their Arduino shield:
https://en.robotis.com/shop_en/item.php?it_id=902-0146-000

Does not compile for the new UNO R4 boards.

More details on the forum posting.

To reproduce, install those libraries using the library manager. 
Load one of the examples, I used the scan_dynamixel.ino.

And try compile.
```
FQBN: arduino:renesas_uno:minima
Using board 'minima' from platform in folder: C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2
Using core 'arduino' from platform in folder: C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2

Detecting libraries used...
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\FD507D8329F1DFE2A4230DE8826AA014/scan_dynamixel.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/defines.txt" -DCFG_TUSB_MCU=OPT_MCU_RAXXX "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2/cores/arduino/tinyusb" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA" "-iprefixC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2" "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/includes.txt" "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\FD507D8329F1DFE2A4230DE8826AA014\\sketch\\scan_dynamixel.ino.cpp" -o nul
Alternatives for DynamixelShield.h: [DynamixelShield@0.2.6]
ResolveLibrary(DynamixelShield.h)
  -> candidates: [DynamixelShield@0.2.6]
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\FD507D8329F1DFE2A4230DE8826AA014/scan_dynamixel.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/defines.txt" -DCFG_TUSB_MCU=OPT_MCU_RAXXX "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2/cores/arduino/tinyusb" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\DynamixelShield\\src" "-iprefixC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2" "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/includes.txt" "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\FD507D8329F1DFE2A4230DE8826AA014\\sketch\\scan_dynamixel.ino.cpp" -o nul
Alternatives for Dynamixel2Arduino.h: [Dynamixel2Arduino@0.6.3]
ResolveLibrary(Dynamixel2Arduino.h)
  -> candidates: [Dynamixel2Arduino@0.6.3]
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\FD507D8329F1DFE2A4230DE8826AA014/scan_dynamixel.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/defines.txt" -DCFG_TUSB_MCU=OPT_MCU_RAXXX "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2/cores/arduino/tinyusb" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\DynamixelShield\\src" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\Dynamixel2Arduino\\src" "-iprefixC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2" "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/includes.txt" "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\FD507D8329F1DFE2A4230DE8826AA014\\sketch\\scan_dynamixel.ino.cpp" -o nul
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\FD507D8329F1DFE2A4230DE8826AA014/scan_dynamixel.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/defines.txt" -DCFG_TUSB_MCU=OPT_MCU_RAXXX "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2/cores/arduino/tinyusb" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\DynamixelShield\\src" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\Dynamixel2Arduino\\src" "-iprefixC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2" "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\MINIMA/includes.txt" "c:\\Users\\kurte\\Documents\\Arduino\\libraries\\DynamixelShield\\src\\DynamixelS [... cut ...]
   13  #define TX_LED (BSP_IO_PORT_00_PIN_12)
   14  #define LED_BLUE (BSP_IO_PORT_00_PIN_13)

2 matches across 2 files
```
Issue both with the LED_RED and with LED_BLUE.
These names are used in one of their header files, as part of an enum that is part of a name space.

Suggested Fix: 
Remove these two defines from those two files.  As far as I could tell, these defines were the only place in the cores project that defines or uses these two names.

Removing them allowed the sketch to build and run.

aentinger, 2023-07-24:
I suggest you open an issue with the Dynamixel library, we are likely going to keep those defines.

closed 2023-07-24 by aentinger

KurtE, 2023-07-24:
@aentinger as you suggested:
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/issues/127