arduino/ArduinoCore-renesas #73: Upload to UNO R4 WiFi fails when "1200 bps touch" causes port change

per1234, 2023-07-28:
### Describe the problem

The upload operation for the [**UNO R4 WiFi**](https://docs.arduino.cc/hardware/uno-r4-wifi) follows this sequence:

1. Perform a ["1200 bps touch"](https://arduino.github.io/arduino-cli/dev/platform-specification/#1200-bps-bootloader-reset) to put the board into the mode where it can be flashed.
1. Invoke the `bossac` command that flashes the binary to the board.

It is common for the address of the board port to change after the "1200 bps touch" step and so the Arduino development tools have a system for detecting the post-touch address and using that address when the upload command is generated from [the "pattern"](https://arduino.github.io/arduino-cli/dev/platform-specification/#tools) defined for the board in `platform.txt`. That system involves a post-touch wait to watch for the new port to appear before eventually timing out and resorting to the fallback behavior of using the original address to the upload pattern if no address change was detected. Since some boards will never produce a post-touch address change (meaning the post-touch wait step would only cause an unnecessary delay in the upload operation) it is possible to configure the board to skip the wait step by setting [the `upload.wait_for_upload_port` property](https://arduino.github.io/arduino-cli/dev/platform-specification/#1200-bps-bootloader-reset) to `false` in the board definition.

The assumption was made that the address of the **UNO R4 WiFi** board's port will never experience a post-touch address change, so its `upload.wait_for_upload_port` property is set to `false`. However, the address can change under certain conditions and this causes the `bossac` command invocation to fail with a "`No device found on ...`" error because the original port address is used in the generated command instead of the post-touch address.

### To reproduce

#### Arduino IDE

1. Create a sketch with the following code:
   ```cpp
   #include <HID.h>
   void setup() {}
   void loop() {}
   ```
1. Select the appropriate board and port for the **UNO R4 WiFi** from the Arduino IDE menus.
1. Select **Sketch > Upload** from the Arduino IDE menus.
1. Wait for the upload to complete successfully.
   the upload should be successful as long as the board isn't already running a problematic sketch.
1. Select **Sketch > Upload** from the Arduino IDE menus.

🐛 The upload fails:

```text
Performing 1200-bps touch reset on serial port COM12
No device found on COM12
"C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino5/bossac" -d --port=COM12 -U -e -w "C:\Users\per\AppData\Local\Temp\arduino\sketches\6CF073A8170159518E82E009B3D511D0/sketch_jul27a.ino.bin" -R
Failed uploading: uploading error: exit status 1
```

#### Arduino CLI

##### Setup

```text
$ arduino-cli version

arduino-cli.exe  Version: git-snapshot Commit: f66becdf Date: 2023-07-23T20:34:26Z

$ mkdir /tmp/HIDSketch

$ printf "#include <HID.h>\nvoid setup() {}\nvoid loop() {}\n" > "/tmp/HIDSketch/HIDSketch.ino"

$ arduino-cli compile --fqbn arduino:renesas_uno:unor4wifi /tmp/HIDSketch

[...]

Used library Version Path
HID                  C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\libraries\HID

Used platform       Version Path
arduino:renesas_uno 1.0.2   C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2

$ arduino-cli board list

Port  Protocol Type              Board Name                FQBN                          Core
COM42 serial   Serial Port (USB) Arduino UNO R4 WiFi       arduino:renesas_uno:unor4wifi arduino:renesas_uno

$ arduino-cli upload --fqbn arduino:renesas_uno:unor4wifi --port COM42 --verbose /tmp/HIDSketch  # Upload of the sketch is successful as long as the board is not already running a problematic sketch

Performing 1200-bps touch reset on serial port COM42
"C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino5/bossac" -d --port=COM42 -U -e -w "C:\Users\per\AppData\Local\Temp\arduino\sketches\1F72A0DD582A60EAC64D21C07F93BFD9/HIDSketch.ino.bin" -R
Set binary mode
version()=Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
[... 116 lines cut ...]

https://docs.arduino.cc/tutorials/uno-r4-wifi/usb-hid

The fault will also occur when using either of those libraries in a sketch:

```cpp
#include <Keyboard.h>
void setup() {}
void loop() {}
```

---

Originally reported at https://forum.arduino.cc/t/failed-uploading-after-upload-usb-hid-example/1151778

<a name="workaround"></a>

#### Workaround

1. Connect your Arduino board to your computer with the USB cable.
1. Press and release the button marked "**RESET**" on your Arduino UNO R4 WiFi board quickly twice.
   You should now see the "**L**" LED pulsing.
1. Check the port of the **UNO R4 WiFi** board.
   ❗ The double reset you did in step (1) can cause the port number of the board to change, so don't assume you already have the correct port selected.
1. Upload the sketch to the board as usual.

mjs513, 2023-07-28:
@per1234 - you might want to see this and the following posts: https://github.com/arduino/ArduinoCore-renesas/issues/62#issuecomment-1653725275

per1234, 2023-07-28:
Thanks for your suggestion @mjs513. The fault still occurs when uploading to the UNO R4 WiFi board with the 0.2.1-rc1 version of the firmware on its bridge module.

per1234, 2023-08-22:
A potential fix for this problem has been identified. This involves changes to two software components:

- The "bridge" firmware: https://github.com/arduino/uno-r4-wifi-usb-bridge/issues/13
- The platform: https://github.com/arduino/ArduinoCore-renesas/pull/74

https://github.com/arduino/ArduinoCore-renesas/pull/74 was previously rejected, but the reason for rejecting it (the introduction of a 10 s delay into uploads when the sketch is not using HID) would no longer be valid (at least for users who have the updated firmware) due to the change to the firmware. Those with the older version of the firmware would still be subject to the increased upload durations, but I think this is less problematic when a remedy for the inconvenience is readily available.