arduino/ArduinoCore-renesas #91: R4 Minima: "extra" pins inaccessible.

WestfW, 2023-08-07:
The board to chip pin translations for the RX and TX LEDs, and the RX/TX signals available on the SWD connector are not present in the Minima variants.cpp or pins_arduino.h, making those essentially inaccessible to "proper" arduino sketches.

facchinm, 2023-08-24:
TX/RX LEDS are now user accessible (https://github.com/arduino/ArduinoCore-renesas/blob/main/variants/MINIMA/variant.cpp#L49-L50 , from commit https://github.com/arduino/ArduinoCore-renesas/commit/df98db2d3f9b2b8362b74e9ad720cd13e1a1aac7 ).

For the SWD pins, I don't think it's a good idea to expose them to the end user, but I'll be glad to change my mind if you have a user case for it.

WestfW, 2023-08-24:
I wasn't talking about the SWD pins themselves, but rather the "extra" serial port from P501/502 that seems to be connected pins 7 and 8 on the SWD connector.   (This seems to be an Arduino-specific hack.  Pin 7 is "supposed" to be a "key" (missing pin) for correct connector insertion.  Not that I've ever seen a connector with pin7 actually missing.)

[image]

facchinm, 2023-08-25:
Got it, I agree that's something we should add to the list of user pins (maybe also the SWD ones at this point).
About the "key" pin, the specification only mandates that 7 and 8 are "not connected" on the debugger side 
[image]
 so you can add whatever you want on that pins :slightly_smiling_face:

PR #109 (2023-08-30): Fix pin numbers macros on UNO R4
Fixes #91
closed 2023-08-30 by facchinm