arduino/ArduinoCore-renesas #94: Simple inclusion of <chrono> leads to compilation error

aentinger, 2023-08-09:
* Minimal sketch to reproduce the error:
```C++
#include <chrono>

void setup() {

}

void loop() {

}
```
* Compile the sketch via `arduino-cli compile -b arduino:renesas_portenta:portenta_c33 -v .`.
* Error
```bash
/home/alex/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=200000000 -std=gnu++17 -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"/tmp/arduino/sketches/158652842019AB5611B5802F01A02A76/chrono.ino\"" -DARDUINO_PORTENTA_C33 -DARDUINO_ARCH_RENESAS -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/alex/Arduino/hardware/arduino-git/renesas/variants/PORTENTA_C33/defines.txt -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino/tinyusb -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino/api/deprecated -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino/api/deprecated-avr-comp -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino -I/home/alex/Arduino/hardware/arduino-git/renesas/variants/PORTENTA_C33 -iprefix/home/alex/Arduino/hardware/arduino-git/renesas @/home/alex/Arduino/hardware/arduino-git/renesas/variants/PORTENTA_C33/includes.txt /tmp/arduino/sketches/158652842019AB5611B5802F01A02A76/sketch/chrono.ino.cpp -o /dev/null
Error while detecting libraries included by /tmp/arduino/sketches/158652842019AB5611B5802F01A02A76/sketch/chrono.ino.cpp
Generating function prototypes...
/home/alex/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=200000000 -std=gnu++17 -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"/tmp/arduino/sketches/158652842019AB5611B5802F01A02A76/chrono.ino\"" -DARDUINO_PORTENTA_C33 -DARDUINO_ARCH_RENESAS -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/alex/Arduino/hardware/arduino-git/renesas/variants/PORTENTA_C33/defines.txt -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino/tinyusb -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino/api/deprecated -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino/api/deprecated-avr-comp -I/home/alex/Arduino/hardware/arduino-git/renesas/cores/arduino -I/home/alex/Arduino/hardware/arduino-git/renesas/variants/PORTENTA_C33 -iprefix/home/alex/Arduino/hardware/arduino-git/renesas @/home/alex/Arduino/hardware/arduino-git/renesas/variants/PORTENTA_C33/includes.txt /tmp/arduino/sketches/158652842019AB5611B5802F01A02A76/sketch/chrono.ino.cpp -o /tmp/arduino/sketches/158652842019AB5611B5802F01A02A76/preproc/sketch_merged.cpp
In file included from /tmp/chrono/chrono.ino:1:0:
/home/alex/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:266:38: error: macro "abs" passed 2 arguments, but takes just 1
       abs(duration<_Rep, _Period> __d)
```

Workaround:
```diff
-#include <chrono>
+#undef abs
+#include <chrono>
```

facchinm, 2023-08-24:
That patch was inherited by https://github.com/arduino/ArduinoCore-mbed/commit/2042db15a90339714f0654416f26c6e389c71aba , which unfortunately doesn't link the underlying issue :disappointed:

AgainPsychoX, 2024-03-12:
I wish it was fixed, it's driving me crazy....