arduino/ArduinoCore-renesas #135: UNOWIFOR4 - Sketch that uses HID - Some Serial.print(...) don't output.

KurtE, 2023-09-15:
The file <HID.h> remaps Serial to USBSerial.
This only works for those files that include HID.h

**Problem/Example:**

So for example if you have a sketch like:
```
extern void print_without_hid(int loop_count);
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while(!Serial && millis() < 5000) {}
  delay(1000);

}

int loop_count = 0;
void loop() {
  Serial.print("Loop Count: ");
  Serial.print(++loop_count, DEC);
  Serial.print(" Baud: ");
  //Serial.println(Serial.baud(), DEC);
  Serial.println();
  print_without_hid(loop_count);
  delay(1000);
}
```
And a second tab, with name like: not_including_hid.cpp
```
#include <Arduino.h>
void print_without_hid(int loop_count) {
  Serial.print("\tNO HID: ");
  Serial.println(loop_count, DEC);
}
```
The prints in this second file will not show up in the Serial monitor.  As they instead are output through _UART1_ to the ESP32-S3.

**Possible Solutions:**

1) Create a new implementation for Serial that detects or is told which of the two paths to take:  Example from a test sketch I have
```
class _Serial : public _SerialUSB {
public:
  _Serial() {}
  void begin(unsigned long baud = 115200) {
    if (_native_USB) SerialUSB.begin(baud);
    else _UART1_.begin(baud);
  }
  void begin(unsigned long baud, uint16_t config) {
    if (_native_USB) SerialUSB.begin(baud, config);
    else _UART1_.begin(baud, config);
  };
  void end() {
    if (_native_USB) SerialUSB.end();
    else _UART1_.end();
  }

  void nativeUSB(bool native_usb) {
    if (native_usb != _native_USB) {
      _native_USB = native_usb;
      if (native_usb) {
        pinMode(21, OUTPUT);
        digitalWrite(21, HIGH);
        __USBStart();
        SerialUSB.begin(115200);
      } else {
        pinMode(21, OUTPUT);
        digitalWrite(21, LOW);
        // need some way to tell SerialUSB to end and other to start
        SerialUSB.end();
      }
    }
  }

  virtual int peek() {
    if (_native_USB) return SerialUSB.peek();
    else return _UART1_.peek();
  }
  virtual int read() {
    if (_native_USB) return SerialUSB.read();
    else return _UART1_.read();
  }
  virtual int available() {
    if (_native_USB) return SerialUSB.available();
    else return _UART1_.available();
  }
  virtual int availableForWrite() {
    if (_native_USB) return SerialUSB.availableForWrite();
    else return _UART1_.availableForWrite();
  }
  virtual void flush() {
    if (_native_USB) SerialUSB.flush();
    else _UART1_.flush();
  }

  virtual size_t write(uint8_t c) {
    if (_native_USB) return SerialUSB.write(c);
    else return _UART1_.write(c);
  }
  virtual size_t write(const uint8_t *p, size_t len) {
    if (_native_USB) return SerialUSB.write(p, len);
    else return _UART1_.write(p, len);
  }

  using Print::write;

  operator bool()  {
    if (_native_USB) return SerialUSB;
    else return _UART1_;

  }

private:
  bool _native_USB = false;
};

```

2) Remap in Arduino.h
Like:
```
#ifndef NO_USB
#define Serial  SerialUSB
#define Serial1 _UART1_
#define Serial2 _UART2_
#define Serial3 _UART3_
#define Serial4 _UART4_
#define Serial5 _UART5_
#else
#if __has_include(<HID.h>)
#define Serial SerialUSB
#else
#define Serial _UART1_
#endif
#define Serial1 _UART2_
#define Serial2 _UART3_
#define Serial3 _UART4_
#define Serial4 _UART5_
#endif

```
Which worked with my example.  

2a) same as 2, but maybe has:
```
#if __has_include(<HID.h>)
#define USE_USBSERIAL_FOR_SERIAL
#endif
...
#else
#ifdef USE_USBSERIAL_FOR_SERIAL
#define Serial SerialUSB
#else
#define Serial _UART1_
#endif
```
Which also allows you to define it some other ways, like in boards.txt...

Thoughts?  Preferences?

facchinm, 2023-09-16:
Hi Kurt,
the second option wouldn't work since the `Serial*` macros are part of the core folder that's untouched by user includes. So `__has_include(HID)` would always be false. A better option would probably be to undef Serial and redefine inside the HID library (but only for UNO R4 WiFi and not for Minima).

In any case, HID on UNO R4 WiFi is not as easy to get right as in other boards with Native USB, so documenting properly is very important (and the Serial object mapping could definitely be documented better)
@alranel @karlsoderby

KurtE, 2023-09-16:
Thanks @facchinm,
>The second option wouldn't work since the Serial* macros are part of the core folder that's untouched by user includes. So
> __has_include(HID) would always be false. A better option would probably be to undef Serial and redefine inside the HID >library (but only for UNO R4 WiFi and not for Minima).

As you mentioned (or implied), there is no fully clean answer.  My thoughts, which could easily be wrong was the __has_include(HID) will likely always be false when compiling the core, as you hopefully cached the builds in the  %temp%/arduino/cores/(core name).  And I have not looked to see if the compile of this is influenced at all by the sketch that it is compiling.  That is if the first time I compile for WIFI without changes in core is using HID, will that be picked up.... Maybe I should just look to see if the -I for the HID library is included on those lines that compile the core.

But where hopefully it does work, is for the libraries or other source code (.cpp files) in the sketch directory, Will then have the Serial remapping done.  

I included the simple test sketch I mentioned above.
[test_sketch_with_hid-230916a.zip](https://github.com/arduino/ArduinoCore-renesas/files/12640753/test_sketch_with_hid-230916a.zip)

>In any case, HID on UNO R4 WiFi is not as easy to get right as in other boards with Native USB, so documenting properly is >very important (and the Serial object mapping could definitely be documented better)
Side note: Still not sure why by default the ESP32-S3 owns the USB versus the RA4M1, but that is a different topic.

HID.h alread has: 
```

#ifdef ARDUINO_UNOWIFIR4
#undef Serial
#define Serial SerialUSB
#endif
```

But this only works for the source files that specifically include HID.h.   So if the sketch that does HID, includes any other libraries, they typically do not include this file, So if they do any Serial outputs, it will go to Serial which is mapped to _UART1_, which does not make it out through USB.

facchinm, 2023-09-16:
Ah, indeed you are very right! The undef/define would work in combination with `__has_include` (if placed in `Arduino.h`) for every external library since they are treated as standalone compilation units. 
It wouldn't work inside the core but it's not a real issue.
Anyway, just for the sake of completeness, for library writers I always suggest to add logging or debugging via a [`setDebug()` function](https://github.com/arduino-libraries/ArduinoBLE/blob/dff9e414abc20df491b46d0b38b360eb1a3507fb/src/utility/HCI.cpp#L661) that accepts a `Stream&` object and must be called from the user code, thus removing all the possible bugs due to slightly different implementations.

KurtE, 2023-09-18:
@karlsoderby  - Sorry the Zip file had the #include <HID.h>
No try uncommenting it and you will see the NO HID line will no longer output.

Also you will probably find that at least on Windows, it will show up as a different COM port, and it may be 
difficult to then reprogram the board.   Usually took a couple of double-clicks of reset and/or maybe it was a single reset followed quickly by a double reset, that then you hear windows chime of USB something changed.

karlsoderby, 2023-09-18:
Strangely for me, the `NO HID` line does actually print even without an inclusion of the `<HID.h>` library. Can't work out why.

Regardless, I am in the process of adding the current state of the implementation to the documentation, which is:
- Including `<HID.h`> enables a new set of methods (`baud()`, `stopbits()` etc.)
- This is due to the remapping of the `Serial` object to the `SerialUSB` object that is defined in the `HID.h` file.
- Added a note for USB port issue. This has occured to me as well, but not very frequent (using MacOS M1). 

Thank you @KurtE for taking the time to report the issues in such detail.