arduino/ArduinoCore-renesas #149: [Feature Request] Provide a much more up-to-date compiler

aentinger, 2023-09-25:
This core (as well as many other Arduino cores) ships with  GCC `7.2.1 20170904`. 

```bash
~/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
```

Meanwhile `arduino-pico` uses GCC 12 and [stm32duino](https://github.com/stm32duino/Arduino_Core_STM32) uses GCC 10 (at least).

I suggest updating tooling to GCC 12.

Fashion-Corp, 2023-09-25:
On a serious note, Alexander, what is the benefit of these continuous
updates of everything? It can have many benefits using a stable, older
version of everything.

aentinger, 2023-09-26:
ArduinoCore-renesas leverages C++17 - which is only incompletely supported by GCC 7.