arduino/ArduinoCore-renesas #158: Regarding Serial[x].flush()

TNobuhara, 2023-10-05:
Using Serial1.flush()
In flush() of ArduinoUnoR3 waited until the transmission was completely finished.
When I tried R4, it seemed to exit as soon as the Tx buffer was empty.

Please tell me what to do. Or is it possible to fix it?

delta-G, 2024-05-06:
Serial on the R4 doesn't currently use the `txBuffer` at all.  `Serial.write` is blocking by itself which makes 'flush' redundant.  I've submitted PR #304 to modify the `UART` class to use the buffer and not to block.  This by itself restores the function of flush.  It will still block until the buffer is empty, but the buffer won't be empty until the transmission end interrupt at the end of the transmission.

per1234, 2024-11-30:
Closing as resolved by https://github.com/arduino/ArduinoCore-renesas/pull/405

closed 2024-11-30 by per1234