arduino/ArduinoCore-renesas #160: P214 and P215 problem

simooon442, 2023-10-10:
Hello.
On a custom minima pcb, I'm trying to read P214 and P215 with this code 
`#include <Arduino.h>
#define INPUT_PIN BSP_IO_PORT_02_PIN_15
void setup() {
  R_IOPORT_PinCfg(NULL, INPUT_PIN, IOPORT_CFG_PORT_DIRECTION_INPUT);
  Serial.begin(115200);
}

void loop() {
  bsp_io_level_t ret;
  R_IOPORT_PinRead(NULL, INPUT_PIN, &ret);
  if(ret == BSP_IO_LEVEL_LOW) {
    Serial.println("Pin LOW");
  } else {
    Serial.println("Pin HIGH");
  }
  delay(500);
}`

But without success, ports state is only **LOW**, unable to read high level on these ports.

But when I try to read P204 for example, then I'm able to read different level state.

My question is, is the minima bootloader take into account these two port ? P214 and 215.

Thanks

guurii, 2023-10-27:
I am also facing the same problem. Did you find any solution?

simooon442, 2023-10-30:
> I am also facing the same problem. Did you find any solution?

Hello, we solve the problem by setting parameter called "BSP_CLOCK_CFG_SUBCLOCK_POPULATED" to 0 in the bsp_cfg.h

maxgerhardt, 2023-10-30:
What magic is it that turning "subclock populated" to 0 (I guess that means "no") makes specific GPIO pins work, when different GPIO pins worked fine before this? Do you have a reference to the relevant sections in the chip's manual?

facchinm, 2023-10-30:
Pins P214 and P215 are supposed to be connected to a 32k quartz in case BSP_CLOCK_CFG_SUBCLOCK_POPULATED is set to 1; the mux is probably going to prevent the pins reconfiguration afterwards (the reference manual doesn't report anything on this topic but it looks like a sensible choice)

paulvha, 2023-10-31:
it is explained and validated at https://forum.arduino.cc/t/custom-pcb-p214-port-reading/1176644

closed 2023-11-13 by facchinm