arduino/ArduinoCore-renesas #162: UNO R4 CAN - Mask & Filter

Woodwarth, 2023-10-13:
There is no possibulity to set CAN Mask and Filter

aentinger, 2023-10-16:
You are correct. I believe what you want to say is: "Feature Request: please add possibility to configure CAN filtering", right?

Woodwarth, 2023-10-31:
new post in forum with working solution

aentinger, 2023-11-02:
Great! How about providing a [pull request](https://docs.github.com/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request) so that we can add your changes to this repository? I can assist you in case there are any problems 😉 .

Woodwarth, 2023-11-02:
Hello,Never done this bevore, if you can say me how to do this, ill do it. 👌

per1234, 2025-11-17:
Closing as resolved by https://github.com/arduino/ArduinoCore-renesas/pull/170.

Thanks @Woodwarth!

closed 2025-11-17 by per1234