arduino/ArduinoCore-renesas #163: [UNO R4 MINIMA] error code of  CAN.write() in Arduino_CAN Libraries

JpOKAYU, 2023-10-14:
What does the error code in the return value of CAN.write(msg) mean?
For example, I got the following error message from `rc = CAN.write(msg)` when I ran the sample program, "CANWrite.ino".
What does this mean?
`
CAN.write(...) failed with error code -60003
`

aentinger, 2023-10-16:
The error you see is this one: [`FSP_ERR_CAN_TRANSMIT_NOT_READY`](https://github.com/arduino/fsp/blob/8ec537bc86b1b71203d72d0cbb28e6e75b353ebd/ra/fsp/inc/fsp_common_api.h#L286).

Very likely this means that you have no termination and/or incorrect bus setup.

[This forum article](https://forum.arduino.cc/t/using-can-with-minima-r4/1162333/13?u=lxrobotics) should be able to help you find approaches to debug your problem.

Final Note: GitHub issues is not for support requests, that's what the forum is for. A GitHub issue is for bug reports or feature requests.

closed 2023-10-16 by aentinger

JpOKAYU, 2023-10-17:
Thank you for your thoughtful answer!
I will be careful how I use issues.