arduino/ArduinoCore-renesas #171: [Portenta C33] IIC communication failed

boschLining, 2023-11-03:
I want to use Portenta C33 to communicate with my slave device, so I powered it up by using USB and connect 3V3 PIN for powering my slave device, what's more, my SCL(PIN 12) SDA(PIN 11) had been connected correctly, but I still had a issue "Read failed with error code: 5"
The following is my codes, regarding reading chip id of my slave device.
`#include <Wire.h>

void setup() {
  Serial.begin(9600);
  Wire.begin();
}

void loop() {
  byte error, data;
  int addr = 0x76;
  int reg = 0xd0;

  Wire.beginTransmission(addr);
  Wire.write(reg);
  error = Wire.endTransmission();

  if (error == 0) {
    Wire.requestFrom(addr, 1);
    if (Wire.available()) {
      data = Wire.read();
      Serial.print("Data read: ");
      Serial.println(data);
      Serial.println("Read success");
    }
  } else {
    Serial.print("Read failed with error code: ");
    Serial.println(error);
  }

  delay(1000);
}
`
So do you have any ideas about it, do I need to modify any source codes, liking set PINS or set CLOCK?
Thanks.

boschLining, 2023-11-03:
And I catch the singal, you can see the VDD is normal, but there is no CLK from the master, do you have any files to enable/disable IIC?

boschLining, 2023-11-03:
I don't know why I can not upload my picture on github, may be a issue. But, there is no clock from SCL and no data from SDA on saleae

aentinger, 2023-11-03:
>I don't know why I can not upload my picture on github, may be a issue.

Just use drag-and-drop. That usually works fine.

Is your I2C/Wire slave address correct? What's the device? Are the wires connected correctly? Do you have pull-ups?

I may also note, that the issue tracker is for bugs with the ArduinoCore, not with providing support. That's better done in the [forum](https://forum.arduino.cc/). I leave this issue open a bit more in case there might a real bug, but I rate that probability as low.

boschLining, 2023-11-03:
Hi aentinger,
Thanks for your reply, after pulling up, I can use IIC successfully.
What's more, why I didn't pull up with other Arduino boards, which can run normally, but for Portenta C33, I need to pull up IIC.

closed 2023-11-03 by boschLining

aentinger, 2023-11-03:
I2C requires pull-ups. That you didn't need them before means they were somewhere else in the I2C wiring. But I'm glad you got your issue fixed :+1: