arduino/ArduinoCore-renesas #180: Arduino UNO R4 and Wire library

Pascal2739, 2023-11-08:
Hello, for several years, I have been working with UNO R3 cards linked together by an I2C bus. Each of these cards are used alternately as a “Master” component and a “Slave” component. In my sketch, the Wire library is declared "slave" by the Wire.begin(address) instruction and everything works without a hitch. Recently, I purchased a UNO R4 card and it too will be integrated into my I2C network. On the I2C bus, a DS1307 real-time clock is installed which works perfectly with UNO R3 or MEGA cards. However, on the R4, access to the DS1307 only works if the card is declared as a master component exclusively (Wire.begin()). I tried using the Wire.end() function, but it had no effect. Does anyone have an idea to resolve this problem? THANKS

aentinger, 2023-11-08:
Hi @maidnl :coffee: :wave: 

Can you please take a look at this?

maidnl, 2023-11-10:
@Pascal2739 @aentinger 
The PR https://github.com/arduino/ArduinoCore-renesas/pull/182 should solve this problem (it was not possible to switch role on I2C bus). Please let give a try and let me know. Thanks.

Pascal2739, 2023-11-10:
Thank you, however, having no skills in using GITHUB (this is my first time using it), what should I do with the pieces of code? Do they replace lines of code in the library?. I'm really sorry I don't understand, but usually I never modify libraries. Thank you again for your attention.

maidnl, 2023-11-10:
@Pascal2739 no problem at all. Just wait then... the fix will be evaluated by my colleagues and if it works as expected it will be integrated (this however will take some times, so be patient). At the end you will see in the IDE you are using that an update is available, at this point if you install the update your R4 will (hopefully) work as the R3.

Pascal2739, 2023-11-10:
Thank you, I will wait, my software is under development and can wait. Thank you for everything.

closed 2023-11-16 by per1234