arduino/ArduinoCore-renesas #181: RTC battery supply switch disabled

simooon442, 2023-11-08:
Hello.
After each pcb RST (POR or not), the RTC is resetted, because VBTCR1.BPWSWSTP bit is set to 1 after each pcb reset (renesas datasheet section 11.2.1).

I suspect Arduino settings to automatically set this bit at each pcb power restart. How to deactivate it so we can use the RTC function normally ?

We need this VBTCR1.BPWSWSTP bit to be setting at 0 when PCB powering on, otherwise, RTC function is unusable.

gpb01, 2024-03-09:
Try adding the following lines immediately after setup():

```
R_SYSTEM->PRCR    = 0xA502;
R_SYSTEM->VBTCR2  = 0x80;
R_SYSTEM->VBTCR1  = 0x00;
R_SYSTEM->PRCR    = 0xA500;
```

... the battery should now be working ;-)

Guglielmo

simooon442, 2024-07-11:
Hello all.
Thank you to @gpb01 and @andreagilardoni for these helpul work !

@alranel , unfortunately, I gave up since I got no response for months... I don't have anymore my customs chield for trying that.
I used an external RTC instead.

This renesas rtc seems to be very touchy, so I wonder if its a good idea to use it for a reliable project.

alranel, 2024-08-22:
Fixed as of core 1.2.1

closed 2024-08-22 by alranel