arduino/ArduinoCore-renesas #202: WiFi SSID with space does not work

adgbu, 2023-12-11:
Unable to connect to a WiFi network with a name (SSID) that contains a space (e.g. "My wifi").

This happens on an Arduino Uno R4 WiFi board. The WiFi firmware is updated to the current latest (0.3.0). The problem can be replicated with the examples under WiFiS3 from the Arduino IDE, e.g. WiFiUdpSendReceiveString.ino.

I have confirmed that the same sketch does connect to WiFi without problem if I just use a different local WiFi which has a name (SSID) without spaces.

In both cases the security setting is a WPA2 password and the password contains only ASCII chars 33-126.

It makes no difference if the sketch includes <WiFiS3.h> or <WiFi.h>.

I am pretty sure that I have connected to the same network (e.g. "My wifi") before with Arduino framework sketches running on both ESP32 DevKitC and on Raspberry Pi Pico W. I believe this is an issue specifically with ArduinoCore-renesas.

adgbu, 2023-12-12:
I have now confirmed that the same sketch works on the Raspberry Pi Pico W board and connects to "My wifi" without problem. Programmed from Arduino IDE 2.2.1 using the Arduino-Pico core by [earlphilhower](https://github.com/earlephilhower/arduino-pico).

QuintonBonnett, 2025-05-12:
I'm also having this issue! Any chance we can get a bugfix for this? In the meantime, I'll change my SSID to not include spaces.

andreagilardoni, 2025-05-13:
The version `0.3.0` is quite old, can you try with newer ones? With version `0.5.2` everything seems to work correctly. You can follow [this guide](https://support.arduino.cc/hc/en-us/articles/9670986058780-Update-the-connectivity-module-firmware-on-UNO-R4-WiFi)