arduino/ArduinoCore-renesas #230: Optimize for Debugging compiles with `-0s`

zfields, 2024-01-05:
### Describe the problem

When I select "Optimize for Debugging" in the "Sketch" menu, it introduces the `-g3` flag, which is great, but it also leaves `-Os` (optimize for size).

[image]

Normally debug binaries are compiled with `-O0` or `-Og`, to prevent variables or instructions from being optimized away. This allows the executed instructions to resemble the code as much as possible and facilitates debugging.

```
FQBN: arduino:renesas_portenta:portenta_c33
Using board 'portenta_c33' from platform in folder: /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5
Using core 'arduino' from platform in folder: /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5

Detecting libraries used...
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=200000000 -std=gnu++17 -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 -DPROJECT_NAME="/tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino" -DARDUINO_PORTENTA_C33 -DARDUINO_ARCH_RENESAS_PORTENTA -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/defines.txt -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/tinyusb -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated-avr-comp -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33 -iprefix/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5 @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes.txt /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/sketch/BlinkDebug.ino.cpp -o /dev/null
Generating function prototypes...
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=200000000 -std=gnu++17 -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 -DPROJECT_NAME="/tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino" -DARDUINO_PORTENTA_C33 -DARDUINO_ARCH_RENESAS_PORTENTA -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/defines.txt -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/tinyusb -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated-avr-comp -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33 -iprefix/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5 @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes.txt /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/sketch/BlinkDebug.ino.cpp -o /tmp/3961272034/sketch_merged.cpp
/home/zak/.arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/3961272034/sketch_merged.cpp
Compiling sketch...
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -Wall -Wextra -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -MMD -nostdlib -DF_CPU=200000000 -MMD -std=gnu++17 -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -DARDUINO=10607 "-DPROJECT_NAME=\"/tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino\"" -DARDUINO_PORTENTA_C33 -DARDUINO_ARCH_RENESAS_PORTENTA -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/defines.txt -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/tinyusb -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated-avr-comp -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33 -iprefix/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5 @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes.txt /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/sketch/BlinkDebug.ino.cpp -o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/sketch/BlinkDebug.ino.cpp.o
Compiling libraries...
Compiling core...
Using previously compiled file: /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/tmp_gen_c_files/common_data.c.o
Using previously compiled file: /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/tmp_gen_c_files/pin_data.c.o
Using previously compiled file: /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/tmp_gen_c_files/main.c.o
Using previously compiled file: /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/variant.cpp.o
Using precompiled core: /tmp/arduino/cores/arduino_renesas_portenta_portenta_c33_6f630dca8d5eb29b7ef91b56c46bda6c/core.a
Linking everything together...
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -Wl,--gc-sections --specs=nosys.specs -Wall -Wextra -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.elf -L/tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA -L/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33 -T/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/fsp.ld /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/sketch/BlinkDebug.ino.cpp.o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/tmp_gen_c_files/common_data.c.o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/tmp_gen_c_files/main.c.o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/tmp_gen_c_files/pin_data.c.o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/variant.cpp.o -Wl,--whole-archive -Wl,--start-group /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/libs/libfsp.a /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/../../cores/arduino_renesas_portenta_portenta_c33_6f630dca8d5eb29b7ef91b56c46bda6c/core.a -Wl,--no-whole-archive -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group -Wl,-Map,/tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.map
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-objcopy -O binary -j .text -j .data /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.elf /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.bin
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-objcopy -O ihex -j .text -j .data /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.elf /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.hex

/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-size -A /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.elf
Sketch uses 94332 bytes (4%) of program storage space. Maximum is 2097152 bytes.
Global variables use 20720 bytes (3%) of dynamic memory, leaving 502904 bytes for local variables. Maximum is 523624 bytes.
```

[... cut ...]
I would expect to see `-O0` or `-Og` used to compile the program. 

### Arduino IDE version

Version: 2.2.1 Date: 2023-08-31T14:16:33.536Z CLI Version: 0.34.0  Copyright © 2024 Arduino SA

### Operating system

Linux

### Operating system version

Ubuntu

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-01-05:
Hi @zfields. The [`compiler.optimization_flags.debug` property](https://arduino.github.io/arduino-cli/latest/platform-specification/#optimization-level-for-debugging) has not been configured in the "Arduino Renesas Portenta Boards". This means the **Sketch > Optimize for Debugging** option in Arduino IDE has no effect. The compilation will proceed as appropriate for production application regardless of the setting of that menu option.

A `compiler.optimization_flags.debug` property in this platform would not be directly useful to Arduino IDE users because the platform has not been configured for use with the [Arduino IDE 2.x integrated sketch debugger](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-debugger). I suppose it might be useful for those who are using an external debugger though.

zfields, 2024-01-05:
Yes, I do have an external debugger.

It is odd that the "professional" version of your hardware does not support debugging.

As a workaround, is there a way I can manually update the settings file, so that I can enable debugging and move forward?

closed 2024-01-15 by aentinger

zfields, 2024-01-25:
Thanks for getting this fix in!

How do I get access to the fix?

Is there a new version of the Renesas core I need to download, or do I need to pull the nightly build of the IDE v2.0 to get this update?

zfields, 2024-01-25:
I've updated my file locally, so there is no urgency, but I'd love a more formal solution.

aentinger, 2024-01-26:
You'll get a new version for downloading and installing both via IDE and `arduino-cli` once a new version of this core is released. For reference take a look at [arduino-cli core update-index](https://arduino.github.io/arduino-cli/0.35/commands/arduino-cli_core_update-index/) and [arduino-cli core upgrade](https://arduino.github.io/arduino-cli/0.35/commands/arduino-cli_core_upgrade/), i.e.

```bash
arduino-cli core update-index
arduino-cli core upgrade arduino:renesas_uno
```

If/when a new core is going to be published is a question I can not answer as its subject to many internal considerations and dependencies - sorry.

zfields, 2024-01-26:
Thank you for providing those resources and your candor.

PR #238 (2024-01-15): Define the `compiler.optimization_flags` allows for specifying the desired optimization level during compilation.
This fixes #230 .

This change can be verified by examining the verbose compilation output (`-v`) when compiling a sketch **with** (`--optimize-for-debug`) and **without** optimization for debug turned on.

```bash
arduino-cli compile -b arduino-git:renesas:unor4wifi -v MySketch --optimize-for-debug
```

```bash
arduino-cli compile -b arduino-git:renesas:unor4wifi -v MySketch
```