arduino/ArduinoCore-renesas #235: SE05X library update from NXP

jowin202, 2024-01-10:
Hello,

I would like to ask, if there is a possibility to exchange the current "middleware" of SE05X with the newest version of provided by NXP.

I would like to have the NXP version (3886 lines of code):
https://github.com/NXP/plug-and-trust/blob/master/hostlib/hostLib/se05x_03_xx_xx/se05x_APDU_impl.h

instead of the current Arduino version (1073 lines of code):
https://github.com/arduino/ArduinoCore-renesas/blob/main/libraries/SE05X/src/lib/apdu/se05x_APDU_impl.c

this is the same file but with different versions. The NXP version supports RSA encryption and HMAC, while the Arduino version does not support RSA (!)

pennam, 2024-01-10:
Hi @jowin202 
this is not a newest version, but a different version. The SE05X library is based on the NXP  PlugNTrust nano package https://github.com/NXPPlugNTrust/nano-package 

If you need more APDUs you can port them from the full PlugNTrust library, but if I recall correctly you need to slightly adapt them to work and not only copy-paste the functions.

jowin202, 2024-01-10:
So there won’t be an official port to arduino? :((

I’ll try to do it myself, but I actually bought the Portenta for RSA and HMAC. Is there any recommendations where to begin?

pennam, 2024-01-12:
nothing planned yet as far as i know, but i will be happy to help if you want to contribute.

I suggest you to start identifying the new APDU functions that you need from the PlugNTrust library.

Then if you compare our APDU file against the NXP nano one you will notice that we have already added some functions like:

`Se05x_API_GetRandom`
`Se05x_API_DigestInit`
`Se05x_API_DigestUpdate`
....

If you take this functions (from libraries/SE05X/src/lib/apdu/se05x_APDU_impl.c) and compare them to the full package (plug-and-trust/hostlib/hostLib/se05x_03_xx_xx/se05x_APDU_impl.h) you  should be able to see what you need to change to let them build:

[image]