arduino/ArduinoCore-renesas #240: Lock in C33 using ethernet adapter and TCP connection in a long run

Eduardo-Lopes, 2024-01-15:
Hey all.
I implemented a HTTP server on C33.
But in a stress test, the board doesn't do well.
I take the smallest sample.
```
#include <EthernetC33.h>

EthernetServer server(7);

void setup() {
  Serial.begin(9600);

  if (Ethernet.begin() == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
  }
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
}

void loop() {
  EthernetClient client = server.available();
  if (client) {
    char previous_c = 0;
    int sequential_newline = 0;
    char buffer[1024];
    int pos = 0;

    while (client.connected()) {
      if (client.available()) {
        char c = client.read();

        if (previous_c == '\r' && c == '\n')
          sequential_newline++;
        else if (previous_c != '\n' || c != '\r')
          sequential_newline = 0;

        if (sequential_newline == 2)
          break;

        previous_c = c;
        buffer[pos++] = c;
      }
    }
    
    client.write(buffer, pos );

    delay(1);
    // close the connection:
    client.stop();
  }
}
```

and the python code to call
```
import socket

server_address = ('192.168.0.103', 7)  # Replace with the desired server address and port

while True:
    tcp_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    tcp_socket.connect(server_address)

    data = b"POST /mode/IDLE HTTP/1.1\r\nHost: 192.168.0.103\r\nUser-Agent: python-requests/2.28.1\r\nAccept-Encoding: gzip, deflate, br\r\nAccept: */*\r\nConnection: keep-alive\r\nContent-Length: 0\r\n\r\n"
    tcp_socket.sendall(data)

    while True:
        response = tcp_socket.recv(1024)

        if not response:
            break

        print(response.decode(), end='')

    tcp_socket.close()
```

I take a deep look at the source code too. It looks like a problem in lwip function
tcp_output that doesn't clean the recv buffer in some way.
And it looks like that is something low level because the C33 stops responding to ping too.
In lwip functions is called tcp_write to add the buffer and tcp_output to send the buffer, but tcp_write keeps returning ERR_MEM.

Also, there is no SDCard on the shield.

Any help?

Eduardo-Lopes, 2024-01-15:
Adding...
This code uses the lwip directly and responds the same way.

```
#include <EthernetC33.h>

#include "lwipServer.h"
#include "EthernetClient.h"

void tcpecho_raw_init(void);
err_t tcpecho_raw_accept(void *arg, struct tcp_pcb *newpcb, err_t err);
err_t tcpecho_raw_sent(void *arg, struct tcp_pcb *tpcb, u16_t len);

static struct tcp_pcb *tcpecho_raw_pcb;

bool restart = true;

enum tcpecho_raw_states
{
  ES_NONE = 0,
  ES_ACCEPTED,
  ES_RECEIVED,
  ES_CLOSING
};

struct tcpecho_raw_state
{
  u8_t state;
  u8_t retries;
  struct tcp_pcb *pcb;
  /* pbuf (chain) to recycle */
  struct pbuf *p;
};

void
tcpecho_raw_free(struct tcpecho_raw_state *es)
{
  Serial.println("raw_free");

  if (es != NULL) {
    if (es->p) {
      /* free the buffer chain if present */
      pbuf_free(es->p);
    }

    mem_free(es);
  }  
}

void
tcpecho_raw_close(struct tcp_pcb *tpcb, struct tcpecho_raw_state *es)
{
  Serial.println("raw_close");

  tcp_arg(tpcb, NULL);
  tcp_sent(tpcb, NULL);
  tcp_recv(tpcb, NULL);
  tcp_err(tpcb, NULL);
  tcp_poll(tpcb, NULL, 0);

  tcpecho_raw_free(es);

  tcp_close(tpcb);

  restart = true;
}

void
tcpecho_raw_send(struct tcp_pcb *tpcb, struct tcpecho_raw_state *es)
{
[... 256 lines cut ...]
  } else {
    /* abort? output diagnostic? */
  }
}

void setup()
{
  if (Ethernet.begin() == 0) {
    Serial.println("Failed to set static IP");
  }

  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
}

void loop()
{
  if (restart)
  {
    restart = false;
    tcpecho_raw_init();
  }
  delay(10);
}
```

facchinm, 2024-01-16:
@Eduardo-Lopes can you try applying this PR https://github.com/arduino/ArduinoCore-renesas/pull/234 ? It should improve things drastically. /cc @andreagilardoni

andreagilardoni, 2024-01-17:
@Eduardo-Lopes As I remember I encountered this issue while dealing with #234, it has to be something related to missing deallocation of pbuf. If you use #234 this should be fixed. That PR should be able to handle higher loads with ethernet, I will also provide some docs on how to tweak performances to reach the highest possible performances, if you need that now I can provide that to you.

Eduardo-Lopes, 2024-01-17:
Thanks! I was trying to compile the PR with e2 Studio, but no success.

If there is a way that you can help me.

My case doesn't involve high performance, but reliability.

andreagilardoni, 2024-01-17:
I don't know how to help you with e2 Studio, you can have a try with arduino IDE or arduino-cli. 

As described, #234 also addresses reliability.