arduino/ArduinoCore-renesas #249: Blink application with `-Wpedantic` flag produces myriad warnings

zfields, 2024-01-25:
I added `-Wpedantic` to the `platform.txt` file, and there are too many warnings to paste in this issue.

These should be addressed to ensure the stability of the platform.

```
FQBN: arduino:renesas_portenta:portenta_c33
Using board 'portenta_c33' from platform in folder: /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5
Using core 'arduino' from platform in folder: /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5

Detecting libraries used...
...
Generating function prototypes...
...
Compiling sketch...
...
Compiling libraries...
Compiling core...
...
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -Wall -Wextra -Wpedantic -Og -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -MMD -nostdlib -DF_CPU=200000000 -MMD -std=gnu++17 -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -DARDUINO=10607 "-DPROJECT_NAME=\"/tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino\"" -DARDUINO_PORTENTA_C33 -DARDUINO_ARCH_RENESAS_PORTENTA -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/defines.txt -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/tinyusb -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated-avr-comp -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33 -iprefix/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5 @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes.txt /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/variant.cpp -o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/variant.cpp.o
In file included from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/instances/r_lpm.h:34:0,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra_gen/hal_data.h:30,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/tmp_gen_c_files/main.c:2:
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/api/r_lpm_api.h:97:41: warning: ISO C restricts enumerator values to range of 'int' [-Wpedantic]
     LPM_SNOOZE_REQUEST_AGT3_UNDERFLOW = 0x100000000ULL, ///< Enable AGT3 underflow snooze request
                                         ^~~~~~~~~~~~~~
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/api/r_lpm_api.h:98:41: warning: ISO C restricts enumerator values to range of 'int' [-Wpedantic]
     LPM_SNOOZE_REQUEST_AGT3_COMPARE_A = 0x200000000ULL, ///< Enable AGT3 compare match A snooze request
                                         ^~~~~~~~~~~~~~
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/api/r_lpm_api.h:99:41: warning: ISO C restricts enumerator values to range of 'int' [-Wpedantic]
     LPM_SNOOZE_REQUEST_AGT3_COMPARE_B = 0x400000000ULL, ///< Enable AGT3 compare match B snooze request
                                         ^~~~~~~~~~~~~~
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/api/r_lpm_api.h:184:39: warning: ISO C restricts enumerator values to range of 'int' [-Wpedantic]
     LPM_STANDBY_WAKE_SOURCE_IIC0    = 0x80000000ULL,  ///< I2C 0 interrupt
                                       ^~~~~~~~~~~~~
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/api/r_lpm_api.h:185:39: warning: ISO C restricts enumerator values to range of 'int' [-Wpedantic]
     LPM_STANDBY_WAKE_SOURCE_AGT3UD  = 0x100000000ULL, ///< AGT3 underflow interrupt
                                       ^~~~~~~~~~~~~~
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/api/r_lpm_api.h:186:39: warning: ISO C restricts enumerator values to range of 'int' [-Wpedantic]
     LPM_STANDBY_WAKE_SOURCE_AGT3CA  = 0x200000000ULL, ///< AGT3 compare match A interrupt
                                       ^~~~~~~~~~~~~~
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes/ra/fsp/inc/api/r_lpm_api.h:187:39: warning: ISO C restricts enumerator values to range of 'int' [-Wpedantic]
     LPM_STANDBY_WAKE_SOURCE_AGT3CB  = 0x400000000ULL, ///< AGT3 compare match B interrupt
                                       ^~~~~~~~~~~~~~
In file included from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/PluggableUSB.h:23:0,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/ArduinoAPI.h:33,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/Arduino.h:4,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/variant.cpp:1:
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/USBAPI.h:38:5: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
     };
     ^
In file included from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/ArduinoAPI.h:54:0,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/Arduino.h:4,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/variant.cpp:1:
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/Compat.h:8:2: warning: extra ';' [-Wpedantic]
 };
  ^
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/Compat.h:12:2: warning: extra ';' [-Wpedantic]
 };
  ^
In file included from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/Arduino.h:16:0,
                 from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/variant.cpp:1:
/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/usb/SerialUSB.h:77:2: warning: extra ';' [-Wpedantic]
 };
  ^
...
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gcc -c -Wall -Wextra -Wpedantic -Og -g3 -nostdlib -DF_CPU=200000000 -MMD -std=gnu11 -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -DARDUINO=10607 -DARDUINO_PORTENTA_C33 -DARDUINO_ARCH_RENESAS_PORTENTA -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/defines.txt -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/tinyusb -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/deprecated-avr-comp -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino -I/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33 -iprefix/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5 @/home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/variants/PORTENTA_C33/includes.txt -o /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/core/cortex_handlers.c.o /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/cortex_handlers.c
In file included from /home/zak/.arduino15/packages/arduino/hardware/renesas_portenta/1.0.5/cores/arduino/api/PluggableUSB.cpp:20:0:
[... cut ...]
/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-objcopy -O ihex -j .text -j .data /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.elf /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.hex

/home/zak/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-size -A /tmp/arduino/sketches/6E18701710D06A6EAAFD2F664971D8EA/BlinkDebug.ino.elf
Sketch uses 103852 bytes (4%) of program storage space. Maximum is 2097152 bytes.
Global variables use 20736 bytes (3%) of dynamic memory, leaving 502888 bytes for local variables. Maximum is 523624 bytes.
```

aentinger, 2024-01-26:
Well, in such a large application consisting of quite some 3rd party code this is a bit of a pointless exercise. If you take a look at the error messages you'll see that quite some originate i.e. from the Renesas FSP layer. They would have to be fixed there first. As for `-Wpedantic` errors originating from code under our control please feel free to provide a PR to address them. Just understand that if you enable `-Wpedantic -Werror` there will very likely always be failing code pieces due to 3rd party software.

zfields, 2024-01-26:
I'm not trying to point the finger at anyone, or make "pedantic" warnings seem super important. Furthermore, I fully understand that `-Wpedantic` is an extreme configuration, and I understand many of them have to do with third-party code. That being said, it's is still worth cleaning up the things that can be cleaned, and may be under the scope of your control.

I'm more than happy to contribute and I will try to create PRs as I can :+1:, but to provide completely candid feedback -- my urge is to switch platforms entirely. I offer this only to say, when other people see such a significant amount of warnings generated, they may have a similar reaction.

Now that I know you are open to it, I will try to send PRs your way.

aentinger, 2024-01-29:
>Now that I know you are open to it, I will try to send PRs your way.

Absolutely. Looking forward to getting those PRs :rocket: