arduino/ArduinoCore-renesas #251: WiFi.ping() function on WiFiS3 library doesn't exist

gpb01, 2024-01-29:
The WiFi.ping() function, that exists in the library [WiFiNINA](https://github.com/arduino-libraries/WiFiNINA/blob/12c33f10cc3eddad245a2263fe0476fee60320a2/src/WiFi.cpp#L379), **doesn't exist** in the WiFiS3 library and there is no way to ICMP ping network devices.

Can you please add such a function?

Thanks in advance,

Guglielmo Braguglia

TeubyDE, 2024-02-13:
would be really nice to have this feature!

paulvha, 2024-02-14:
I have taken some time to include PING. It requires updates in both WiFiS3 as well as the UNOR4USBBridge-software. In the attached document I have documented the changes to be made.
[add_PING.docx](https://github.com/arduino/ArduinoCore-renesas/files/14284289/add_PING.docx)

paulvha, 2024-02-14:
I have spend time to add PING. it requires changes to both the UNOR4Bridge software as well as the WiFiS3 library. In the attached zip is a document with the changes to be applied. Also the files that need changes are included for reference.
[WiFiPing.zip](https://github.com/arduino/ArduinoCore-renesas/files/14284384/WiFiPing.zip)

regards,
Paul

gpb01, 2024-02-14:
> In the attached zip is a document with the changes to be applied.

Hi Paul,
thanks a lot for your work, but ... if you don't create a "_fork_" and do an official "_pull request_" ... I don't know if the Arduino team will consider it ... :-\ 

Guglielmo

paulvha, 2024-02-14:
hi Guglielmo,
I had considered doing that, but I am not sure the Arduino Team is interested. Several other pending bridge issues could be solved and some don't look that difficult to solve. I was doing this exercise to understand how the bridge software is working. It took longer time to get started as good documentation was missing and different links & scripts were outdated. 

regards,
Paul

JAndrassy, 2024-02-14:
@paulvha they are interested. I have now 23 PR merged in core and 3 in firmware

paulvha, 2024-02-14:
I can make a fork and pull request if interested.  I also saw noticed update to the partition table for the ESP32 (i used the latest version from yesterday). I have made a pull request for the firmware and will add one for the core.  Did you also solve the getTime() issue 206?

regards
Paul

privitera, 2024-06-03:
Wondering, what is the status of this?

voidzero, 2024-06-19:
I just got my first R4 and am very excited to try this out. Thanks for the pull request, hope it will be accepted.

PR #395 (2024-11-27): Add WiFi.ping()
Fixes #251
closed 2024-11-27 by pennam