arduino/ArduinoCore-renesas #265: Support the Mutual TLS Authentication on WiFiS3 library

ArduinoGetStarted, 2024-02-15:
Hello the DEV team,

Do you have a plan to support the mutual TLS authentication on the WiFiS3 library ?

This feature is available on ESP32 now. The Uno R4 uses the ESP32 as WIFI module, So I think it does not need to develop the from scratch.

pennam, 2024-02-15:
Hi @ArduinoGetStarted,
If I am correct this should be a matter of adding two new AT commands for this APIs

```
setCertificate(const char *client_ca);
setPrivateKey (const char *private_key);
```

ArduinoGetStarted, 2024-02-15:
Dear @pennam ,

I think you are correct. The `WiFiClientSecure` library from ESP32 has those function. I tested, it works well with AWS IoT.
The link for the tutorial is here: [how to connect ESP32 to AWS IoT](https://esp32io.com/tutorials/esp32-aws-iot).
I hope this feature will be available on Arduino Uno R4 soon!
Uno form factor is compatible with many shields

bayu01, 2024-03-13:
Being new to the Arduino development I naively thought the Uno R4 Wifi would have the latest and greatest functionality. I stumbled into this issue trying to follow the most basic AWS IoT Core documentation. 

Is there no current workarounds?

ArduinoGetStarted, 2024-03-14:
Dear @bayu01 
I think the Uno R4 WiFi is not mature. It is new released. It will be a good choice for every application for several years later.
For now, you can use Arduino Nano ESP3. You can see the tutorial for [Arduino Nano ESP32 connects to AWS IoT](https://newbiely.com/tutorials/arduino-nano-esp32/arduino-nano-esp32-aws-iot)