arduino/ArduinoCore-renesas #274: Building libfsp.a fails

paulvha, 2024-02-25:
In the README.MD are the instructions to rebuild the libfsp.a
Unfortunately, the content for extras/e2studioProjects/Santiago / UNOR4 WiFi does not work. 

- The configuration is for the 48-pin R7FA4M1AB3CNE and  NOT for the 64-pin R7FA4M1AB3CFM Renasas chip.
- The configuration.xml content is incomplete as it relates to the Interrupts and pins

Next to the MINIMA information is also missing.

Can you update the information.

regards,
Paul

bplu4t2f, 2024-04-10:
Additionally, I'm getting this error:

```
Building file: ../ra/fsp/src/rm_vee_flash/rm_vee_flash.c
../ra_gen/hal_data.c:366:76: error: 'VECTOR_NUMBER_KEY_INT' undeclared here (not in a function); did you mean 'VECTOR_NUMBER_AGT0_INT'?
  366 | { .channel_mask = 0, .trigger = KEYMATRIX_TRIG_RISING, .ipl = (12), .irq = VECTOR_NUMBER_KEY_INT, .p_callback =
```

This is with FSP version 4.0.0, which should be the correct version I believe.
I tried updating to FSP 5.2.0 (current version) just to see how it goes, but that caused even more problems, so I'm trying to reproduce the 4.0.0 build first.

paulvha, 2024-04-10:
I had the same errors...the information is incomplete..

facchinm, 2024-04-23:
@bplu4t2f the error while building the project via e2studio is expected; in fact, the particular architecture of interrupts in RA chips makes it impossible to allocate all interrupts in one single binary. This doesn't prevent however a successful rebuild of libfsp after running [fsp_to_arduino.sh](https://github.com/arduino/ArduinoCore-renesas/blob/main/extras/e2studioProjects/Santiago/fsp_to_arduino.sh)
@paulvha the project / configuration is shared between Minima and WiFi and is not intended to be complete (pinmuxing is wrong etc) since we only need it to produce the makefile and the correct defines to then build libfsp

closed 2024-04-23 by facchinm

facchinm, 2024-04-24:
@bplu4t2f sorry to hear it didn't work :cry:  The script is indeed supposed to work out of the box from a clean repo comed with `--recursive` (so fsp version is the expected one)  after the e2studio project has been build once (with or without errors, not important) to create the makefile and the defines.

bplu4t2f, 2024-04-24:
@facchinm For example, the `arm` directory in `.../fsp_patched/ra/` (and therefore also `/extra/fsp`) is missing. `board`, `fsp` and `tes` are there. I would guess that I somehow have wrong submodule commit checked out, but I just double checked it.
Not sure what's going on, but I suppose it is a problem on my end somehow, so thanks for the confirmation.

paulvha, 2024-04-26:
Hi

I have spend more time yesterday and today to create libfsp.a on my Ubuntu system and I am (finally) able to create a libfsp.a without errors during  fsp_to_arduino.sh.. (size is 10.5Mb)

However when I compile a simple sketch that just prints "hallo" every second it compiles and uploads without any problem. BUT.... it does NOT work with this new libfsp.a, nothing is displayed or happening, even after pressing reset or removing and re-applying power. Using the same sketch but now with the libfsap.a that is part of the 1.1.0 version, it works as expected.

Below are my steps: What am I missing or doing wrong ?

creating libfsp.a for R4-WIFI

get an updated fork from https://github.com/arduino/ArduinoCore-renesas
Copy to another location to test. Wait for the copy to finish.

OBTAIN THE SUBMODULES:
======================
git submodule update --force --recursive --init --remote --no-single-branch
This must complete without errors

CREATE THE NECESSARY FILES:
=============================

Get & Start E2 (2023-4 / FSP version 4.4.0)
create a new workspace
Import an existing project from ../ArduinoCore-renesas/extras/e2studioProjects/Santiago
double click configuration.xml: it will now update the files from 4.0.0 to 4.4.0

  *
the submodule extras/fsp that is downloaded and used for fsp_fatched is 4.4.0
  *
The configuation.xml in Santiago (and other) still refers to 4.0.0 (Is autodated file the root cause ??)

Build for Debug (click on hamer in the top row)
The build will fail (or maybe not), it does not matter. it will generate the necessary files.

BUILD LIBFSP.A
==============
cd ../ArduinoCore-renesas/extras/e2studioProjects/Santiago
./fsp_to_arduino.sh
This must complete without errors and the libfsp.a is copied into the variants-folder for the UNOR4WIF/libs.
In case there is an error that BSP_CLOCKS_PLL_DIV_12 has not been defined. I had the wrong FSP version in E2 to create ra_gen.

Instead of E2 buld for debug, I tried build for release, but then got error message from fsp_to arduino.sh :./Debug/makefile: No such file or directory

Any advice is welcome..

regards,
Paul

paulvha, 2024-04-28:
hi

FINALLY some success and got "something" to work with version 4.0.0.

Mixing the configuration files in ArduinoCore-renesas/extras/e2studioProjects/Santiago (based on 4.0.0) and the Arduino core files with the 4.4.0 DOES NOT work.

The submodule extras/fsp that is downloaded and used for fsp_fatched is 4.4.0 BUT should be 4.0.0.

I followed the steps as indicated in the earlier mail below, but now copied the whole UNOR4WIFI variant folder. HOWEVER cores/Arduino/time.cpp failed to compile as the Asynchronous timer memory variables definition in the BSP file is different than what time.cpp/ micros() is expecting. 4.4.0 has a new different structure : R_AGTX0 instead of R_AGT0. Even after making that change it compiled but did not work. I wonder what other changes have been made.

CAN YOU POINT TO THE PLACE WHERE THE 4.0.0 PATCHED LIBRARY IS?

Using the following steps I can create a working (but not in-depth tested) variant UNOR4WIFI :

get an updated fork from https://github.com/arduino/ArduinoCore-renesas
Copy to another location to test. Wait for copy to finish.

OBTAIN THE SUBMODULES:
======================
git submodule update --force --recursive --init --remote --no-single-branch

CREATE THE NECESSARY FILES:
=============================
Start E2 (2022-7 / FSP version 4.0.0)
create a new workspace
import existing project from ArduinoCore-renesas/extras/e2studioProjects/Santiago
Build for debug (click on hamer in the top row)
The build will fail (or maybe not), it does not matter. it will generate the necessary files.

BUILD LIBFSP.A with the standard 4.0.0
==============
cd ArduinoCore-renesas/extras/e2studioProjects/Santiago
edit fsp_to_arduino.sh and COMMENT OUT line 22 : find ra -type f | xargs -I{} cp fsp_patched/{} {}
save it
./fsp_to_arduino.sh

As you compile you might get warnings... ignore those

MAKE THE RIGHT COPY.
====================
cd ArduinoCore-renesas/variants folder that has been updated with this build
copy UNOR4WIFI
goto the IDE folder packages/arduino/hardware/renesas_uno/1.1.0/variants
rename the current (original UNOR4WIFI) to UNOR4WIFI.org for a safety copy
paste the new/updated UNOR4WIFI now.

Regards,
Paul