arduino/ArduinoCore-renesas #277: [Feature Request] Core version access in a header file

lyusupov, 2024-02-29:
Other Arduino Cores, such as:
1) ESP32 ( [core_version.h](https://github.com/espressif/arduino-esp32/blob/2fdd90163936c95cef3cd28dbddee5ccc3275f4d/.github/scripts/on-release.sh#L240) )
2) RP2040 (  [Version.h](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/RP2040Version.h)  )
3) STM32 ( [def.h](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/cores/arduino/stm32/stm32_def.h)  )

\- all they use a definition of the Core Release Version located in a header file.

If would be nice to have this useful feature in **ArduinoCore-renesas** as well.

aentinger, 2024-03-01:
Sure, would you like to prepare a PR for this?

lyusupov, 2024-03-01:
The project maintainer is the person who knows the best way to implement this request.

aentinger, 2024-03-01:
The project maintainers may be otherwise occupied, since this is open-source and its something that you are interested in having why not go ahead and provide a suggestion?