arduino/ArduinoCore-renesas #301: .gitmodules is pointing to an incorrect version of fsp source code

paulvha, 2024-05-03:
With reference to issue #274 I have spent a lot of time getting the right source code to compile a working libfsp.a.  During these tests I have tried to follow the short description in the README file.

There were several issues to overcome, but the biggest is that the submodule link for extras/fsp in .gitmodules is pointing to [https://github.com/arduino/fsp](https://github.com/arduino/fsp). That place contains version V4.4.0 of the FSP source code and this code does not work with the rest of the source code. 

However when following online from [ArduinoCore-renesas/extras ](https://github.com/arduino/ArduinoCore-renesas) the link in the '**extras/fsp & 5df91**', you will get redirected to https://github.com/arduino/fsp/tree/5d13f916c72f9d1471864361dfaf8f492cab5979. THAT PLACE DOES CONTAIN  V4.0.0 and the latest patches.

By manually downloading this V4.0.0 in 'extras/fsp' of the local copy of the source code and then compiling, it generates a working libfsp.a as expected.

facchinm, 2024-05-03:
@paulvha the commit you linked is just 3 commits over 4.0.0 (see https://github.com/arduino/fsp/commits/5d13f916c72f9d1471864361dfaf8f492cab5979/ ), so the gitmodule is ok

paulvha, 2024-05-03:
unfortunately, it is not working. I have tried it many many many times. You can compile the libfsp.a without problems, but there are 2 issues :

V 4.4.0 has made change to AGT structure definition and as such cores/Arduino/time.cpp failed to compile as the Asynchronous timer memory variables definition in the BSP file is different than what time.cpp/ micros() is expecting. 4.4.0 has a new different structure : R_AGTX0 instead of R_AGT0. This was also raised by a different FSP user  as an issues for V4.3.0

Even after making that change it compiled but did not work. I wonder what other changes have been made.  

regards,
Paul

facchinm, 2024-05-03:
I think I didn't express myself properly, the version linked by gitmodules is NOT 4.4.0 or 4.3.0, the repo contains them but when properly cloned via `git submodule init && git submodule update` it will checkout 4.0.0 with 3 patches applied on top.

paulvha, 2024-05-03:
Thanks

I will test it now..
regards,
Paul

paulvha, 2024-05-03:
works.. now I think the 'no-single-branch' that was part of the git submodule update was causing the error