arduino/ArduinoCore-renesas #327: CAN Transmitter:  error -60003.

AndreaZGuz, 2024-05-28:
Hello there!

We are trying to reproduce the communication in [this tutorial](https://docs.arduino.cc/tutorials/uno-r4-minima/can/). We are struggling with this error shown in the output console for the "transmitter" device: `CAN.write(...) failed with error code -60003`.
A little more context:

- We are using a TI TCAN1051HGV transceiver. This works fine if used with Arduino Giga R1 WiFi. As we know, Giga R1 operating voltage is 3.3V (in this case, we use 5V to supply the transceiver and also 3.3V to its level shifter, so it can work with the logic operating voltage for Giga R1). Datasheet specifies that if using a MCU working on 5V, is not necessary to use the level shifter. However, we tried both ways (3.3V to level shifter or not connecting this to any voltage).
- Oscilloscope draws the train of pulses for the CAN TX pin (D4 on Minima version, D10 on WiFi version), but not for the CAN RX pin (like the communication is not being completed with the transceiver).
- Adding to the above point and according to the core files, the error received: [FSP_ERR_CAN_TRANSMIT_NOT_READY](https://github.com/arduino/fsp/blob/8ec537bc86b1b71203d72d0cbb28e6e75b353ebd/ra/fsp/inc/fsp_common_api.h#L286) means that the transmission is in progress (probably, the last frame was not sent).
- No CAN frames are shown in the oscilloscope if we test CANH and CANL buses.
- We tried to communicate Giga R1 (Sender) and Arduino UNO R4 Minima/WiFi (receiver). Using and not using terminating resistors on the receiver side. The performance is the same. Giga is properly sending data but Arduino UNO R4 is not processing the information.
- Baudrate was always the same for both, sender and receiver. We tried every available option (from 250K to 1000K). 
- PINS for CAN_TX and CAN_RX for both, R4 Minima and R4 WiFi are properly set in their core and Arduino_CAN library.
- Following the process suggested [here](https://forum.arduino.cc/t/using-can-with-minima-r4/1162333/18) clears the error but UNO R4 is still not able to send data through CANH and CANL. CANTX and CANRX both show the train of pulses (just because they are wired together).

Any insights about the origin and how to solve this will be appreciated. Thank you!

aentinger, 2024-05-29:
Do you have used termination - a 120R resistor at each end of the CAN bus? (Or at the very least a simple 120R resistor between CAN HIGH and CAN LOW?)

aentinger, 2024-05-29:
**Note**: If this does not solve your problem I suggest taking to the [Arduino Forum](https://forum.arduino.cc/) or contact [Arduino support](https://support.arduino.cc/hc/en-us), as this is not a firmware issue.

closed 2024-05-29 by aentinger