arduino/ArduinoCore-renesas #329: Polling rate issues

skhrlx, 2024-06-01:
When using a usb host shield on r4 wifi, my mouse only can get 125hz polling rate (thing that doesn't happens on arduino leonardo with the same code)

I already tested other mouses too, and some friends of mine, it's having the same issues on getting low polling rate

I'm not sure how I can help of provide something, anything you want I can help with it 😃

delta-G, 2024-06-02:
What mouse?  What code?   How did you determine this rate?

skhrlx, 2024-06-02:
For checking current mouse polling rate you can use this site from a mouse provider: https://www.redragon.com.br/pollingrate

You can find some explanation about polling rate on the site above, and a little too here: https://wiki.archlinux.org/title/mouse_polling_rate

_**note: the above site its just for you get what polling rate is, dont mind on os problems or etc...**_

Lower polling rate completely ruins gaming experience and normal use of PC, since an 125hz polling rate have an "considerably" lag compared to an 1000hz polling rate and is visually noticeable

My mouse, don't care here at moment (since i tested with others too...), but it's an "Pichau hive S" with an PMW3389 Sensor Capable of 1000hz polling rate (which works normally when i just plug the mouse on pc, without intercept it with the R4 WIFI + USB Host Shield).

Again, **it's not an problem with my code itself** , since it's working well on my arduino leonardo (works well on R4 Wifi too, i just get this lower polling rate), and it only get this lower polling rate on R4 Wifi, but, on any case, is here my code (a simple mouse parser for the usb host shield + some serial communication that you can completely ignore since its not relevant here): https://github.com/skhrlx/arduino-hostshield

Some other users related this annoying experience here too when using R4 Wifi: https://forum.arduino.cc/t/change-usb-report-rate-on-r4-wifi/1226530

### final note: I'm completely sorry if anything looked if I wrote raging, English is not my language, I'm friendly writing it in the best way I can

skhrlx, 2024-06-02:
I'm autistic and sometimes when I get tired out I completely stop thinking. So, I just read the source again and for my lucky, and good professionals, it is commented out, closing and making a pull request! #331

closed 2024-06-02 by skhrlx