arduino/ArduinoCore-renesas #330: Arduiono_CAN bitrates

leosoftGR, 2024-06-01:
Hello,

Currently, in Arduino_CAN.h are supported only the following bitrates:

[https://github.com/arduino/ArduinoCore-renesas/blob/main/libraries/Arduino_CAN/keywords.txt](url)

BR_100k	LITERAL1
BR_125k	LITERAL1
BR_250k	LITERAL1
BR_500k	LITERAL1
BR_1000k	LITERAL1

Seems that even 100kbit is not yet supported.  However, It would be perfect if you can support 83.3Kbps because it is the standard bitrate of the internal canbus in the most Mercedes Benz cars.  And now if you want to communicate with such models and arduino, you need to use an external module or an add on.  But since UNO R4 wifi has the on board CAN, it would be ideal if you are able to use it directly.

Thanks for your time

per1234, 2024-06-02:
Hi @leosoftGR. Support for setting arbitrary bit rates in addition to the predefined was recently added to the library: https://github.com/arduino/ArduinoCore-renesas/pull/296

I think that capability will be sufficient to satisfy your request so I'll go ahead and close this as resolved.

closed 2024-06-02 by per1234

leosoftGR, 2024-06-02:
> Hi @leosoftGR. Support for setting arbitrary bit rates in addition to the predefined was recently added to the library: #296
> 
> I think that capability will be sufficient to satisfy your request so I'll go ahead and close this as resolved.

Hello,

Is there any documentation or example on how to use arbitrary bitrate values with Arduino_CAN.h ?  Can you send me a link?

Thanks for your time