arduino/ArduinoCore-renesas #341: Linter error 'stdlib.h' file not found clang (pp_file_not_found)

ShortArrow, 2024-06-13:
# Want

Would you tell me which `stdlib.h` are used.
Or, Would fix this linting error.

# Expected

No Lint error.

# Actual

```
In included file: 'stdlib.h' file not found clang (pp_file_not_found) [1, 10]
```

[image]

# PoC

https://github.com/ShortArrow/poc_renesas_uno

# Env

- arduino:renesas_uno:minima
- arduino-cli  Version: 0.35.3-arch Commit: 95cfd654fe0dd6b07e903d3132c0faceabfbe9e

rowingdude, 2024-08-07:
This issue is now two months old, but it's more likely an issue with your environment than an issue with Arduino.

ShortArrow, 2024-08-16:
Fixed by https://github.com/ShortArrow/poc_renesas_uno/commit/3a06f1cc9c83e6fc6094930dd09d33313d7bad6b

closed 2024-08-16 by ShortArrow

realhonbo, 2024-11-01:
I tried download a source file `https://github.com/Patapom/Arduino/blob/master/Libraries/AVR%20Libc/avr-libc-2.0.0/include/stdlib.h` and add it to the clangd file