arduino/ArduinoCore-renesas #371: TextWithArduinoGraphics does not compile due to missing library

sterretjeToo, 2024-09-01:
The example in the subject includes `ArduinoGraphics.h` which is not installed by default. Installing the library using the library manager solves the issue.

```text
/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/arduino-builder -dump-prefs -logger=machine -hardware /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/hardware -hardware /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages -tools /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/tools-builder -tools /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/hardware/tools/avr -tools /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages -built-in-libraries /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/libraries -libraries /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/sketchbook/libraries -fqbn=arduino:renesas_uno:unor4wifi -ide-version=10819 -build-path /tmp/arduino_build_393590 -warnings=all -build-cache /tmp/arduino_cache_403337 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.bossac.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/bossac/1.9.1-arduino5 -prefs=runtime.tools.bossac-1.9.1-arduino5.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/bossac/1.9.1-arduino5 -verbose /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.1.0/libraries/Arduino_LED_Matrix/examples/TextWithArduinoGraphics/TextWithArduinoGraphics.ino
/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/arduino-builder -compile -logger=machine -hardware /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/hardware -hardware /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages -tools /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/tools-builder -tools /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/hardware/tools/avr -tools /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages -built-in-libraries /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/libraries -libraries /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/sketchbook/libraries -fqbn=arduino:renesas_uno:unor4wifi -ide-version=10819 -build-path /tmp/arduino_build_393590 -warnings=all -build-cache /tmp/arduino_cache_403337 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/dfu-util/0.11.0-arduino5 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.bossac.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/bossac/1.9.1-arduino5 -prefs=runtime.tools.bossac-1.9.1-arduino5.path=/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/bossac/1.9.1-arduino5 -verbose /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.1.0/libraries/Arduino_LED_Matrix/examples/TextWithArduinoGraphics/TextWithArduinoGraphics.ino
Using board 'unor4wifi' from platform in folder: /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0
Using core 'arduino' from platform in folder: /home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0
Detecting libraries used...
/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DNO_USB -DBACKTRACE_SUPPORT -DARDUINO_UNOR4_WIFI -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10819 "-DPROJECT_NAME=\"/tmp/arduino_build_393590/TextWithArduinoGraphics.ino\"" -DARDUINO_UNOWIFIR4 -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE=700 -mthumb @/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0/variants/UNOWIFIR4/defines.txt -DCFG_TUSB_MCU=OPT_MCU_RAXXX -I/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/tinyusb -I/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/deprecated -I/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/deprecated-avr-comp -I/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino -I/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0/variants/UNOWIFIR4 -iprefix/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0 @/home/wim/Downloads/1_arduino_ide_1.x/arduino-1.8.19-UnoR4/portable/packages/arduino/hardware/renesas_uno/1.2.0/variants/UNOWIFIR4/includes.txt /tmp/arduino_build_393590/sketch/TextWithArduinoGraphics.ino.cpp -o /dev/null
Alternatives for ArduinoGraphics.h: []
ResolveLibrary(ArduinoGraphics.h)
TextWithArduinoGraphics:2:10: fatal error: ArduinoGraphics.h: No such file or directory
 #include "ArduinoGraphics.h"
  -> candidates: []
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
ArduinoGraphics.h: No such file or directory
```

In my opinion, built-in examples should compile out-of-the-box or should contain sufficient information to get it to compile. Please add a note in the example that this is a requirement.

Reference: Arduino forum topic 1297300 (sorry, no link, from the head forum.arduino.cc/t/1297300).

JAndrassy, 2024-09-01:
install the ArduinoGraphics library in Library Manager in IDE

sterretjeToo, 2024-09-01:
@JAndrassy 

I know and I mentioned that that solves the issue. But not everybody does know; as I mentioned, that example should compile out-of-the-box or dependencies should be made clear.

PR #401 (2024-11-28): TextWithArduinoGraphics: add note about ArduinoGraphics dependency
fixes #371