arduino/ArduinoCore-renesas #406: Non-blocking version of  WiFi::localIP()

ondra-novak, 2024-11-30:
Can we have non-blocking version of the function localIP?

I have WDT set on 5 seconds and this function causes WDT to trigger

https://github.com/arduino/ArduinoCore-renesas/blob/03b4a91db49dae7601701c33cd09924ea6b363d3/libraries/WiFiS3/src/WiFi.cpp#L336

Thank you

JAndrassy, 2024-11-30:
after WiFi.begin, the localIP() should return immediately because WiFi.begin waits for DHCP. how do you initialize WiFi?

ondra-novak, 2024-11-30:
I am also setting timeout to zero, because wifi connection is not mandatory to my device. Its priority is to control connected devices. The control cycle has 1 millisecond resolution. I can handle aprx 20mills a common modem-esp32 exchange, but i cannot handle 5 seconds blocking call.

MYusufY, 2024-12-01:
I made PR that fixes this by adding `localIPNonBlocking()` function.

**PR:**
- #408 

_I hope it fixes your issue and have a nice day :)_

JAndrassy, 2024-12-01:
>  am also setting timeout to zero

but then don't call localIP() if there is no WiFi connoection

ondra-novak, 2024-12-01:
> > am also setting timeout to zero, because wifi connection is not mandatory on my device
> 
> but then don't call localIP() if there is no WiFi connoection

You don't understant. My controller running in the main loop where many operations are scheduler depend on various conditions such a time, sensors or status of various components. One component is Wifi connection. So the task calls WiFi.begin(), then peridodically checks for status and then , when status is OK, retrieves localIP. 

now consider environment with bad wifi connection. The WiFi status is OK, but immediatelly after the wifi connection is lost and localIP blocks. Or other scenario, when WiFi.status is OK, but probably DHCP handshake is still in progress. As I understand, this is the main reason, why the function is blocking. So non-blocking version can simply return "still in progress" and we are good, the Task object can postpone the operation until next cycle.

MYusufY, 2024-12-01:
> > I made PR that fixes this by adding `localIPNonBlocking()` function.
>> 
> > _I hope it fixes your issue and have a nice day :)_
> 
> Thank you, I already happily forked this PR. 

Glad you solved it! Have a nice day.

MYusufY, 2024-12-01:
> correct solution is change the esp32 firmware to not wait for localIP()

You mean using **ESP-IDF**?

But i think solution is solution.

ondra-novak, 2024-12-01:
> I looked more into it. the firmware returns immediately. the localIP() function should not try multiple times. there should not be the loop with a delay. it makes no sense to do that.
> 
> cc: [@pennam](https://github.com/pennam)

Unfortunately, some code expects blocking version of this function, like here
https://github.com/arduino/ArduinoCore-renesas/blob/03b4a91db49dae7601701c33cd09924ea6b363d3/libraries/WiFiS3/src/WiFi.cpp#L183

JAndrassy, 2024-12-01:
> some code expects blocking version of this function, like here

no. that doesn't require a blocking localIP()

ondra-novak, 2024-12-01:
> > some code expects blocking version of this function, like here
> 
> no. that doesn't require a blocking localIP(). config() sets the static IP settings.

I don't know. I assume that setDNS doesn't expect that localIP fails. 

Actually i don't like that code. It is weird. It calls config, however it requires connected wifi. It is really strange. I prefer to set config and then call begin(). And I also missing end() - which I already implemented in my forked version, as calling SOFTRESETWIFI, which should be also available at public API.  There is a lot issues with real usage of Wifi beyond of this discussion. 

(https://github.com/arduino/ArduinoCore-renesas/blob/03b4a91db49dae7601701c33cd09924ea6b363d3/libraries/WiFiS3/src/Modem.cpp#L40) 

by the way, the above code is formally wrong, but luckily works for this specific usage.

pennam, 2024-12-04:
> I looked more into it. the firmware returns immediately. the localIP() function should not try multiple times. there should not be the loop with a delay. it makes no sense to do that.
> 
> cc: [@pennam](https://github.com/pennam)

@JAndrassy  I will look at it for the next release, 1.3.x is already in the oven

PR #450 (2025-03-03): WiFiS3: remove retries to get localIP
Fixes #406
closed 2025-03-03 by pennam