arduino/ArduinoCore-renesas #420: Spin out NXP SE05X library?

jp-bennett, 2024-12-27:
Would you have any interest in spinning the SE05X library out into a separate project? I'm evaluating using an SE050e for Meshtatic, and your little library is far more usable than the official middleware from NXP.

jp-bennett, 2024-12-29:
Just an example of the sort of change I have in mind, it would be great to be able to specify the i2c device in the .begin() call, rather than having the library hard-coded to use Wire3. I'm also planning to add ECDH functions.

pennam, 2025-01-07:
@jp-bennett Thanks for your interest! For your reference the SE05X library is based on this NXP project https://github.com/NXPPlugNTrust/nano-package. Moving out the SE05X library from the core is something i'm thinking about from a while and might be a good idea.