arduino/ArduinoCore-renesas #456: Warnings when compiling 1.4.1 release

jrw429, 2025-03-12:
With the 1.4.1 release there are numerous compilation warnings. I think all of this compiled cleanly in previous release. I'm throwing all this in one report as all showed up together. Some are benign, others could be real bugs - assigning a value to a const, returning a value from a void method, etc.

`.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/libraries/WDT/src/WDT.cpp: In constructor 'WDTimer::WDTimer()':
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/libraries/WDT/src/WDT.cpp:25:14: warning: missing initializer for member 'st_wdt_instance_ctrl::p_context' [-Wmissing-field-initializers]
 , _timeout {0}
              ^
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/libraries/WDT/src/WDT.cpp:25:14: warning: missing initializer for member 'st_wdt_instance_ctrl::p_callback' [-Wmissing-field-initializers]
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/libraries/WDT/src/WDT.cpp:25:14: warning: missing initializer for member 'st_wdt_instance_ctrl::p_callback_memory' [-Wmissing-field-initializers]
`

`.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp: In member function 'bool IRQManager::addPeripheral(Peripheral_t, void*)':
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:385:38: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (irqcfg->first_irq_number != FSP_INVALID_VECTOR) {
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:448:38: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (irqcfg->first_irq_number != FSP_INVALID_VECTOR) {
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:609:26: warning: unused variable 'scfg' [-Wunused-variable]
         i2c_slave_cfg_t *scfg = (i2c_slave_cfg_t *)p_cfg->scfg;
                          ^~~~
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:701:27: warning: unused variable 'mcfg' [-Wunused-variable]
         i2c_master_cfg_t *mcfg = (i2c_master_cfg_t *)p_cfg->mcfg;
                           ^~~~
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp: In member function 'void IRQManager::set_canfd_error_link_event(int, int)':
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:1779:49: warning: unused parameter 'li' [-Wunused-parameter]
 void IRQManager::set_canfd_error_link_event(int li, int ch)
                                                 ^~
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:1779:57: warning: unused parameter 'ch' [-Wunused-parameter]
 void IRQManager::set_canfd_error_link_event(int li, int ch)
                                                         ^~
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp: In member function 'void IRQManager::set_canfd_rx_link_event(int, int)':
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:1794:46: warning: unused parameter 'li' [-Wunused-parameter]
 void IRQManager::set_canfd_rx_link_event(int li, int ch)
                                              ^~
./Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:1794:54: warning: unused parameter 'ch' [-Wunused-parameter]
 void IRQManager::set_canfd_rx_link_event(int li, int ch)
                                                      ^~
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp: In member function 'void IRQManager::set_canfd_tx_link_event(int, int)':
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:1809:46: warning: unused parameter 'li' [-Wunused-parameter]
 void IRQManager::set_canfd_tx_link_event(int li, int ch)
                                              ^~
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/IRQManager.cpp:1809:54: warning: unused parameter 'ch' [-Wunused-parameter]
 void IRQManager::set_canfd_tx_link_event(int li, int ch)
                                                      ^~
`

`.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/cm_backtrace/cm_backtrace.c: In function 'cm_backtrace_init':
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/cm_backtrace/cm_backtrace.c:144:13: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     fw_name = firmware_name;
             ^
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/cm_backtrace/cm_backtrace.c:145:12: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     hw_ver = hardware_ver;
            ^
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/cm_backtrace/cm_backtrace.c:146:12: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     sw_ver = software_ver;
            ^
`

`.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/tinyusb/rusb2/dcd_rusb2.c: In function 'wait_pipe_fifo_empty':
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/tinyusb/rusb2/dcd_rusb2.c:289:3: warning: 'return' with a value, in function returning void
   TU_ASSERT(num);
   ^~~~~~~~~
.../Library/Arduino15/packages/arduino/hardware/renesas_uno/1.4.1/cores/arduino/tinyusb/rusb2/dcd_rusb2.c:288:13: note: declared here
 static void wait_pipe_fifo_empty(rusb2_reg_t* rusb, uint8_t num) {
             ^~~~~~~~~~~~~~~~~~~~
`

pennam, 2025-03-13:
@jrw429 i feel like this warnings were present also in the previous core releases, actually there were many more that have been solved with some of the last PRs. Looking at your log:

* `tinyusb` confirmed and will be fixed with https://github.com/arduino/ArduinoCore-renesas/pull/431
* `cm_backtrace` and `IRQManager` confirmed and will be fixed with https://github.com/arduino/ArduinoCore-renesas/pull/96
* `WDT` confirmed no PR, you can provide one if you wish

jrw429, 2025-03-13:
Thanks for looking at this, and thanks for better formatting the messages. Yes these (and more) may have been present before. The warnings are typically shown once and then cached files are used, so I don't see them often. I will try to look at the WDT warnings, but I won't have time for that for a while.