arduino/ArduinoCore-renesas #463: Arduino UNO R4 SoftwareSerial missing end()

bmitov, 2025-04-26:
The SoftwareSerial implementation for Arduino UNO R4 is missing the end() method.
Sometimes it is necessary to stop the Serial and start manually controlling the pins.
"end()" is implemented for all other platforms that I have worked with :frowning: .

RonsTek, 2025-04-26:
Please fix it!

finnandreh, 2025-04-27:
Hope you can get it fixed quick!

per1234, 2025-04-28:
@bmitov responded to @maidnl's question on a different issue:

> I need to be able to enable the serial, use it as serial, then disable, control the pins, and later enable the serial again.

per1234, 2025-04-28:
I have copied your response from https://github.com/arduino/ArduinoCore-renesas/issues/464 to this issue @bmitov.

I have unlocked the issue in order to allow you to provide any additional information that might be requested. 

However, if the issue becomes the source of "noise" from people making comments that don't add anything of technical value, as happened before, then I will be forced to lock it again. For the interested parties, please refrain from replying unless you have significant and unique information to add. If you only want to express your support for the PR or a comment in the thread, use the ["Reactions" feature](https://github.blog/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to add a "👍", etc. "Me too"/"+1" comments are not allowed here.

Everyone who watches the repository receives a notification whenever any reply is made on any of the issues and PRs. It is important for the developers and maintainers of the project to watch the repository to make sure we see the valuable feedback from the contributors. But if the repository becomes the source of a large number of low value notifications we must either waste our time sorting through them instead of doing productive work, or else unwatch the repository.

Unstructured conversations about Arduino-related subjects are always welcome over on [Arduino Forum](https://forum.arduino.cc/).

pennam, 2025-06-10:
@bmitov @finnandreh @RonsTek  Did you had the chance to test #465 ?

closed 2025-06-30 by facchinm