arduino/ArduinoCore-renesas #466: DFU driver for Windows on ARM

Hsubtnarg, 2025-04-30:
Apologies if the is covered somewhere else, or if this is the wrong place for it.

There is no DFU driver for Windows on ARM, this means that you cannot program the Uno R4 from this platform. Im not sure who needs to write this, but it will become and increasing problem as these computers become more popular ie Microsoft Surface 11 is ARM only.

There have been some discussions on the forums about work around's using things like Zadig and libwdi, but none of them have worked for me. There are others that try to load unsigned drivers but that is not a good long-term solution.
https://forum.arduino.cc/t/windows-on-arm-with-a-qualcomm-x-elite/1278045/2
https://forum.arduino.cc/t/no-duf-capable-usb-device-aviable-r4-minima-cloud-editor-on-surface-pro-x-windows-10-arm-64/1305430
https://github.com/pbatard/libwdi/issues/155#issuecomment-1450642313

I have also looked at Renesas for a solution, but they say they are not supporting Windows on ARM yet
https://community.renesas.com/mcu/tools/f/tools-forum/52448/e2-emulator-drivers-for-windows-11-arm

Just to be clear I am not talking about the IDE, that works fine under emulation. I it is just the DFU driver for the Uno R4 and drivers cannot run in emulation.

facchinm, 2025-05-06:
Hi @Hsubtnarg ,
we've been adapting our driver signing machine for ARM64; unfortunately we can't test as of now, so I'd ask you if you can please try them and see if they work correctly. [renesas.zip](https://github.com/user-attachments/files/20055651/renesas.zip)
To make the installation completely automatic we should also provide the equivalent of [`dpinst`](https://github.com/arduino/ArduinoCore-renesas/tree/main/drivers) for arm64

@lucarin91 @Xayton

Hsubtnarg, 2025-05-06:
Thanks for looking at this. I tried to install the drivers and got this

[cid:f1f896ab-9798-4b16-aa15-65598ea14740]