arduino/ArduinoCore-renesas #487: [Uno R4 WiFi] `while (!Serial);` not working on Uno R4 WiFi

swanjeng, 2025-08-15:
I don't know whether it should be put here or other Github repos about Arduino Uno R4 WiFi. 

## Summary:
On Arduino Uno R4 WiFi, `while (!Serial);` does not block the program until the Serial Monitor is connected, making typical setup() serial prints disappear unless the user manually resets the board. This behavior differs from most other Arduino boards (including Uno R3 and Uno R4 Minima), breaking a large amount of existing example code and tutorials.

## Steps to Reproduce:

1. Upload this sketch to an Arduino Uno R4 WiFi:

```cpp
void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("Hello World");
}
void loop() {

}
```

2. Open Serial Monitor immediately after upload finishes.

3. Observe that no output appears unless the reset button is pressed.

## Expected Behavior:

`while (!Serial);` should block the program until the USB–Serial connection is established, ensuring prints in setup() are visible right after upload, without requiring a manual reset.

## Actual Behavior:

while (!Serial) returns immediately. Output sent before Serial Monitor connection is lost and manual reset is required to see the message.

## Impact:

Breaks compatibility with most Arduino examples and learning materials.

Confuses new users, who expect immediate output after upload.

Makes debugging output in setup() unreliable without additional delay hacks.

Suggested Fixes:

Provide a mechanism for RA4M1 to query USB connection state from the ESP32-S3, making `while (!Serial);` functional.

swanjeng, 2025-08-17:
Here is a video showing the issue.

https://github.com/user-attachments/assets/69ca5798-a70c-4385-a2a4-5bedc84b63dd

code:

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while (!Serial) delay(1);
  Serial.println("Hello~");
  for (int i = 1 ; i <= 200 ; i ++) Serial.println("i = " + String(i));
  pinMode(13, OUTPUT);
  Serial.println("Starts blinking LED");
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

facchinm, 2025-08-21:
Hi @swanjeng ,
the `while (!Serial)` pattern was introduced to allow in fact USB native ports to enumerate before trying to print (on nowhere). On native UART-to-USB like UNO R4 WiFi and UNO R3 the `!Serial` pattern is essentially a no-op. The reason why UNO R3 works is that it gets reset (by design) every time you open the serial port/change the baud rate.
The fix should be applied to the `serial-monitor` project to make it faster at reopening the port after an upload @cmaglie