arduino/ArduinoCore-renesas #490: [Request] Can we have the default factory demo sketch for the Arduino Uno R4 WiFi??

ArsiTheFox, 2025-09-13:
Wondering if we can be given the normal factory demo sketch for the Arduino Uno R4 WiFi??

Would like to have the factory default demo sketch to keep in my files or add Demo files into the board's example sketches

Same with the Arduino Nano R4:
 - https://github.com/arduino/ArduinoCore-renesas/issues/486

per1234, 2025-09-14:
Hi @ArsiTheFox. It is already provided. It is the "MatrixIntro" example sketch of the "LED_Matrix" library that is bundled with the "Arduino UNO R4 Boards" platform installation. You can access it in Arduino IDE by selecting the UNO R4 WiFi board from Arduino IDE's **Tools > Board** menu, then **File > Examples > LED_Matrix > MatrixIntro**.

The sketch is hosted in this repository here:

https://github.com/arduino/ArduinoCore-renesas/blob/main/libraries/Arduino_LED_Matrix/examples/MatrixIntro

closed 2025-09-14 by per1234

per1234, 2025-09-16:
The factory fresh boards have the "Tetris" animation from the example sketch I shared. The "snake" animation you are referring to comes from the "provisioning" program that is automatically uploaded to the board when you set it up as an **Arduino Cloud** IoT [Device](https://docs.arduino.cc/arduino-cloud/hardware/devices). I believe this is the sketch:

https://github.com/arduino-libraries/ArduinoIoTCloud/tree/2.8.0/examples/utility/Provisioning_2.0

The animation is provided by the "Arduino_NetworkConfigurator" library:

https://github.com/arduino-libraries/Arduino_NetworkConfigurator/blob/0.4.1/src/utility/LEDFeedback.cpp#L52

Of course that "provisioning" sketch does much more than only displaying the animation. If you only want to display the "snake" animation, you can use this simple sketch:

```cpp
#include <Arduino_NetworkConfigurator.h>

void setup(){
  LEDFeedbackClass::getInstance().begin();
  LEDFeedbackClass::getInstance().setMode(LEDFeedbackClass::LEDFeedbackMode::BLE_AVAILABLE);
}

void loop() {
  LEDFeedbackClass::getInstance().update();
}
```

If you have any questions about using the sketch, the "Arduino_NetworkConfigurator" library, etc., just post over on Arduino Forum and we'll be happy to assist you there:

https://forum.arduino.cc/